JnyJny/busylight #2: Orjson maturin requirement

Jaharmi, 2020-10-22:
I am unable to install `busylight` on a Raspberry Pi Zero W because the installation fails on `maturin`.

After unsuccessfully trying many other things, I found that this seems to be a requirement of the `orjson` module. It also appears that `orjson` does not currently support 32-bit ARM.

[Raspberry Pi 4 - Cannot install orjson](https://github.com/ijl/orjson/issues/130)

Is `orjson` absolutely required? Could something else be substituted that would work on 32-bit ARM?

JnyJny, 2020-10-22:
I took a look at the dependency tree, `poetry show --tree`, and the package `orjson` is a dependency of `FastAPI`. I've been meaning to refactor `busylight`'s packaging to make the web interface an optional install, so now I've got a good reason to stop putting it off. This assumes that your expected use case does not require the web API.

I don't have an ETA for this fix or a good work around yet.

Jaharmi, 2020-10-22:
I suspected it was something like that.

I would actually like to have Web API access, but I'm not sure if it's required.

My ultimate goal was to try to cobble together something that would work for HomeKit (and/or Home Assistant). I was thinking along the lines of `busylight` -> `homebridge` -> HomeKit (or Home Assistant). I haven't done much at all with Homebridge, so I figured the `busylight` Web API would be another option along with the command line.

JnyJny, 2020-10-22:
I've just published an update, 0.5.8, which breaks out FastAPI and Uvicorn as extra dependencies:

```console
$ pip install busylight-for-humans
...
$ pip list
Package              Version
-------------------- -----------
bitvector-for-humans 0.14.0
busylight-for-humans 0.5.7
click                7.1.2
hidapi               0.9.0.post3
pip                  20.1.1
setuptools           47.1.0
typer                0.3.2
webcolors            1.11.1
$ pip install busylight-for-humans[webapi]
...
$ pip list
Package              Version
-------------------- -----------
bitvector-for-humans 0.14.0
busylight-for-humans 0.5.7
click                7.1.2
fastapi              0.61.1
h11                  0.11.0
hidapi               0.9.0.post3
pip                  20.1.1
pydantic             1.6.1
setuptools           47.1.0
starlette            0.13.6
typer                0.3.2
uvicorn              0.12.2
webcolors            1.11.1
```

It looks like I've managed to remove `orjson` as an indirect dependency, give it a shot on your Raspberry Pi Zero and let me know what breaks.

Jaharmi, 2020-10-23:
That was fast! 😀

Jaharmi, 2020-10-23:
I have the basics working, except for the Luxafor Flag, on 64-bit Raspberry Pi OS on a Raspberry Pi 3 Model B. Now that I've seen it actually control a light, I can try the `busylight` update without FastAPI on my Pi Zero W.

JnyJny, 2020-10-25:
Good to hear it seems to be working for you. Re-open if you continue to have problems.

closed 2020-10-25 by JnyJny