JnyJny/busylight #3: HomeKit Integration

JnyJny, 2020-10-23:
It would be cool have to make the lights controlled by `busylight` be available for use with HomeKit. It appears that that easiest way to make this happen is via the [Homebridge](https://developers.homebridge.io/) project.

- [ ] Write a Homebridge plugin to bridge between busylight web API and HomeKit
- [ ] Update busylight's web API as necessary

Jaharmi, 2020-10-23:
The [Home Assistant](https://www.home-assistant.io) project also allows for HomeKit integration and is Python-based.

I know I've mistakenly created a lot of phantom HomeKit devices with HA, but that's about the extent of what I've done between HA and HomeKit.

JnyJny, 2020-10-23:
I think I'll battle one bear at a time. The last time I played with Home Assistant I did not have a positive experience (trying to connect IKEA smart plugs to HomeKit).

JnyJny, 2020-10-23:
[Busylight Homebridge plugin](https://github.com/JnyJny/busylight-homebridge-plugin)

trahloff, 2024-01-07:
Hi @JnyJny , thank you for creating this amazing project! 

Did your perspective on Home Assistant change over the years? I don't know how it was in 2020, but it became an extremely mature and beloved platform with great support and a very active community. 

I would love to integrate BusyLight for Humans into Home Assistant. Are there any plans to support this?

Jaharmi, 2024-01-07:
@trahloff I've definitely thought about it, as well. I've never done an integration, but I've done a minimal REST setup to at least turn lights on/off and set some basic colors. If you have ideas, I'd be interested in helping somehow.

stuart12, 2025-11-22:
hi @JnyJny thanks for creating busylight-core. It is a good base for adding lights to HomeAssistant. I have added an [Architecture  section](https://github.com/stuart12/busylight-hass?tab=readme-ov-file#architecture) to the README in [busylight-hass](https://github.com/stuart12/busylight-hass) explaining why it would be useful to only enumerate lights with a given path. Let me know if this is not clear.  You can see my deleted attempts to open a light at given USB path in https://github.com/stuart12/busylight-hass/commit/3b6912d2b4a6a6bb289852306a0090c3b9bbe96b. I gave up as I thought that my time was better spent implementing MQTT discovery for [Hass](https://www.home-assistant.io/).

stuart12, 2025-11-22:
[blinkstick-strip](https://www.blinkstick.com/products/blinkstick-strip) lights do work but I sometimes see what I presume is a hardware bug. Sometimes (several times a week), when busylight-core switches the light off, not all of the leds turn off. Some of the top leds (those furthest from the connector) remain on. I will then have to switch the light off many times until fewer leds remain and then finally all leds turn off. I saw the same problem when using the [blinksticklight integration in HomeAssistant](https://www.home-assistant.io/integrations/blinksticklight/) so it is not a bug in [busylight-core](https://github.com/JnyJny/busylight-core/). BTW the HomeAssistant blinksticklight integration was disabled in https://github.com/home-assistant/core/pull/121846 but the documentation was not updated.  If I see this bug too often I will add code to [busylight_hass](https://github.com/stuart12/busylight-hass) to switch the leds off one by one as a possible workaround.

stuart12, 2025-11-23:
thanks @JnyJny `at_path` does the job, see https://github.com/stuart12/busylight-hass/commit/4a04634ea44ed2d2ba2e8c44b1de02ba0ebd7016