JnyJny/busylight #4: Luxafor Flag not recognized or controlled

Jaharmi, 2020-10-23:
I cannot get a Luxafor Flag to be recognized or controlled — at least not as a non-root user.

Steps I took:

1. Installed a 64-bit Raspberry Pi OS on a Raspberry Pi 3 Model B in order to test `busylight`.
2. Installed updates and libraries with `apt-get`:

```
$ sudo apt-get install --upgrade python-dev cython libudev-dev libusb-1.0-0 libusb-dev libusb-1.0-0-dev libhidapi-dev
```

3. Installed `busylight` with `pipenv`.

```
$ pipenv install busylight-for-humans
```

4. Installed the dev-rules and reboot.

```
$ pipenv shell
$ busylight udev-rules -o 99-busylight.rules
$ sudo cp 99-busylight.rules /etc/udev/rules.d
$ sudo reboot
```

5. Found attached lights, which consisted of an Embrava BlyncLight and a Luxafor Flag.

```
$ pipenv shell
$ busylight list --long
ID: Device Name
 0: 
	 0: path: b'0001:0004:00'
	 0: vendor_id: 0x4d8
	 0: product_id: 0xf372
	 0: serial_number: 
	 0: release_number: 0x100
	 0: manufacturer_string: 
	 0: product_string: 
	 0: usage_page: 0x00
	 0: usage: 0x00
	 0: interface_number: 0x00
 1: Blynclight Standard
	 1: path: b'0001:0005:00'
	 1: vendor_id: 0x2c0d
	 1: product_id: 0x0c
	 1: serial_number: 
	 1: release_number: 0x01
	 1: manufacturer_string: 
	 1: product_string: Blynclight Standard
	 1: usage_page: 0x00
	 1: usage: 0x00
	 1: interface_number: 0x00
```

6. Set both lights to red.

```
$ busylight --light-id 0 on red
# The Embrava BlyncLight turned on, set to red color
$ busylight --light-id 1 on red
Light '1' not in the range of 0..0
```

7. Set the Luxafor flag to red.

```
$ sudo -s
$ /home/pi/.local/share/virtualenvs/pi-V9tpwFal/bin/busylight list --long
ID: Device Name
 0: Luxafor Flag
	 0: path: b'0001:0004:00'
	 0: vendor_id: 0x4d8
	 0: product_id: 0xf372
	 0: serial_number: 
	 0: release_number: 0x100
	 0: manufacturer_string: Microchip Technology Inc.
	 0: product_string: LUXAFOR FLAG
	 0: usage_page: 0x00
	 0: usage: 0x00
	 0: interface_number: 0x00
 1: Blynclight Standard
	 1: path: b'0001:0005:00'
	 1: vendor_id: 0x2c0d
	 1: product_id: 0x0c
	 1: serial_number: 
	 1: release_number: 0x01
	 1: manufacturer_string: 
	 1: product_string: Blynclight Standard
	 1: usage_page: 0x00
	 1: usage: 0x00
	 1: interface_number: 0x00
$ /home/pi/.local/share/virtualenvs/pi-V9tpwFal/bin/busylight --light-id 1 off
# The Embrava BlyncLight turned off
$ /home/pi/.local/share/virtualenvs/pi-V9tpwFal/bin/busylight --light-id 0 on red
# The Luxafor Flag turned on, set to red color
```

I'm confused about this outcome. The vendor_id for the Luxafor Flag is in the udev-rules. It seems I should be able to identify and control it with a non-root user in the same manner as the BlyncLight. I can only identify and control the Flag device as root.

JnyJny, 2020-10-23:
I'm trouble shooting on my RPi 3b+ with a Luxafor flag. I *think* it might be a small bug in the generated udev rules file.

JnyJny, 2020-10-23:
Yup. It's a bug in how the udev rules are generated. A USB vendor identifier is a 16-bit quantity, which is generally expressed as a four digit hexadecimal string. The Luxafor vendor identifier is 0x04d8 and when I generate the udev rules, the left-most zero is being truncated from the output.

The udev subsystem must perform a string-wise comparison when it is deciding when to apply a rule, so "04d8" != "4d8" while an integer comparison would succeed.

I'll push a new release in a couple of minutes with the fix included, but you can just edit `/etc/udev/rules.d/99-busylight.rules` to add a zero in front of the "4d8" entries in the file.

JnyJny, 2020-10-23:
Incidentally, when you update the rules it is not necessary to reboot the machine.

```console
$ sudo vi /etc/udev/rules.d/99-busylight.rules
$ sudo udevadm control -R
$ # unplug/plug USB device
$ busylight list
...
```

Jaharmi, 2020-10-23:
This works, if you reboot — and probably _without_ the reboot if you follow _all_ of the instructions and actually unplug/plug the USB device. :)

Thanks!

JnyJny, 2020-10-23:
Thanks for letting me know!

closed 2020-10-23 by JnyJny