JnyJny/busylight #5: BLYNCUSB40S does not `fli`

pcfe, 2020-10-28:
Overall busylight works like a charm with my Embrava Blynclight Plus, except for the Flash Lights Impressively mode.

```bash
[root@workstation ~]# cat /etc/udev/rules.d/99-blynclight-plus.rules 
# BlyncLight Plus, BLYNCUSB40S
SUBSYSTEM=="hidraw*", ATTRS{idVendor}=="2c0d", ATTRS{idProduct}=="0010", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="2c0d", ATTRS{idProduct}=="0010", MODE="0666"
```

```bash
pcfe@workstation ~ $ busylight serve
INFO:     Started server process [55451]
INFO:     Waiting for application startup.
INFO:     Application startup complete.
INFO:     Uvicorn running on http://0.0.0.0:8888 (Press CTRL+C to quit)
INFO:     127.0.0.1:39988 - "GET /1/lights HTTP/1.1" 200 OK
INFO:     127.0.0.1:40004 - "GET /1/light/0/rainbow HTTP/1.1" 200 OK
INFO:     127.0.0.1:40042 - "GET /1/light/0/off HTTP/1.1" 200 OK
INFO:     127.0.0.1:40050 - "GET /1/light/0/on HTTP/1.1" 200 OK
INFO:     127.0.0.1:40054 - "GET /1/light/0/off HTTP/1.1" 200 OK
INFO:     127.0.0.1:40532 - "GET /1/light/0/fli HTTP/1.1" 500 Internal Server Error
ERROR:    Exception in ASGI application
Traceback (most recent call last):
  File "/home/pcfe/.local/lib/python3.9/site-packages/uvicorn/protocols/http/h11_impl.py", line 389, in run_asgi
    result = await app(self.scope, self.receive, self.send)
  File "/home/pcfe/.local/lib/python3.9/site-packages/uvicorn/middleware/proxy_headers.py", line 45, in __call__
    return await self.app(scope, receive, send)
  File "/home/pcfe/.local/lib/python3.9/site-packages/fastapi/applications.py", line 179, in __call__
    await super().__call__(scope, receive, send)
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/applications.py", line 111, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/middleware/errors.py", line 181, in __call__
    raise exc from None
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/middleware/errors.py", line 159, in __call__
    await self.app(scope, receive, _send)
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/middleware/base.py", line 25, in __call__
    response = await self.dispatch_func(request, self.call_next)
  File "/home/pcfe/.local/lib/python3.9/site-packages/busylight/api/__init__.py", line 82, in light_manager_update
    return await call_next(request)
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/middleware/base.py", line 45, in call_next
    task.result()
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/middleware/base.py", line 38, in coro
    await self.app(scope, receive, send)
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/exceptions.py", line 82, in __call__
    raise exc from None
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/exceptions.py", line 71, in __call__
    await self.app(scope, receive, sender)
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/routing.py", line 566, in __call__
    await route.handle(scope, receive, send)
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/routing.py", line 227, in handle
    await self.app(scope, receive, send)
  File "/home/pcfe/.local/lib/python3.9/site-packages/starlette/routing.py", line 41, in app
    response = await func(request)
  File "/home/pcfe/.local/lib/python3.9/site-packages/fastapi/routing.py", line 182, in app
    raw_response = await run_endpoint_function(
  File "/home/pcfe/.local/lib/python3.9/site-packages/fastapi/routing.py", line 133, in run_endpoint_function
    return await dependant.call(**values)
  File "/home/pcfe/.local/lib/python3.9/site-packages/busylight/api/__init__.py", line 366, in Flash_Light_Impressively
    server.manager.apply_effect_to_light(light_id, flash_lights_impressively)
  File "/home/pcfe/.local/lib/python3.9/site-packages/busylight/manager.py", line 295, in apply_effect_to_light
    light.start_effect(effect)
  File "/home/pcfe/.local/lib/python3.9/site-packages/busylight/lights/usblight.py", line 316, in start_effect
    effect(self), f"effect-{self.identifier}"
TypeError: flash_lights_impressively() got an unexpected keyword argument 'color'
```

While I doubt it matters, here's my system info;

```bash
pcfe@workstation ~ $ uname -r                                    
5.8.16-300.fc33.x86_64                                     
pcfe@workstation ~ $ python --version                            
Python 3.9.0                                               
pcfe@workstation ~ $ cat /etc/fedora-release                     
Fedora release 33 (Thirty Three)                           
pcfe@workstation ~ $ python3 -m pip  list --user                                                                             
Package              Version                               
-------------------- -------                               
bitvector-for-humans 0.14.0                                
blynclight           0.12.0                                
busylight-for-humans 0.7.4                                 
click                7.1.2                                 
fastapi              0.61.1                                
h11                  0.11.0                                
loguru               0.5.3                                 
pydantic             1.7                                   
setuptools           50.3.2                                
starlette            0.13.6                                
typer                0.3.2                                 
uvicorn              0.12.2                                
webcolors            1.11.1 
```

As far as I'm concerned, this can be CLOSED WONTFIX. I've used the mode successfully with `blync fli` in the past, but the visual effect is not one I plan to use. But I figured you'd like to know about the color error.

JnyJny, 2020-10-28:
Closed won't fix? LOL. Not on my watch :) Closed, did fix! You've identified a gap in my testing, the web API tests are currently empty. Look for your fix in version 0.7.5 and let me know if you run into any more bugs.

closed 2020-10-28 by JnyJny

pcfe, 2020-10-28:
> Closed won't fix? LOL. Not on my watch :) 
:-)

> Closed, did fix! You've identified a gap in my testing, the web API tests are currently empty. Look for your fix in version 0.7.5 and let me know if you run into any more bugs.

Updated to 0.7.5, I can confirm that `fli` now works.