JnyJny/busylight #10: API server return boolean true if any lights are on or false if all lights are off

Jaharmi, 2020-11-19:
I'm trying to integrate with Home Assistant using the [RESTful Switch](https://www.home-assistant.io/integrations/switch.rest/) platform.

I can't seem to find a way to just get the status of my lights. I'm trying to get the overall on/off status of all connected lights at once. (I can see that having the on/off status of individual lights would be useful, but that's not my current use case.)

The RESTful Switch platform allows accessing one resource/URL to get the state of a "switch" and another to set the state of the "switch." I'm currently able to get my two lights to blink periodically, which I'm guessing is due to some polling.

If the JSON output of something like "/1/lights" included an overall boolean true/false that indicated if any lights were on or if all were off, that seems like it would be pretty easy to interpret with RESTful Switch's [is_on_template](https://www.home-assistant.io/integrations/switch.rest/#is_on_template).

JnyJny, 2020-11-19:
Sure, I can add a "status" entry to the returned JSON payload for the aggregated lights and for individual lights too. I'll add 'color' too. I'll read up more about RESTful Switch, it sounds useful!

JnyJny, 2020-11-19:
Ok it didn't quite make sense to me to modify the `1/lights` end point to include a boolean "any_on" member in addition to the list of per-light descriptions. Each light description JSON payload now has a `is_on` and `color` keys.  The `is_on` key has a boolean value, true if the light is on otherwise false. The `color` key has a string representation of the light's current RGB color in hexadecimal format, e.g. (0xaa,0xbb,0xcc) -> "#aabbcc".  I'd be happy to add another end point just to query any light status if that makes more sense for Switch. Look for the new members in versions past 0.7.7.

JnyJny, 2020-11-19:
Ok, does this make life easier for you? I added a new "lights/status" endpoint that returns the number of the total lights and the number of lights "on". Lights that are illuminated or being actively animated are counted as "on". 

```console
$ curl -s http://localhost:8888/1/lights/status | jq
{
  "number_of_lights": 1,
  "lights_on": 1
}
$ curl -s http://localhost:8888/1/lights/off | jq
{
  "light_id": "all",
  "action": "off"
}
$ curl -s http://localhost:8888/1/lights/status | jq
{
  "number_of_lights": 1,
  "lights_on": 0
}
```

JnyJny, 2020-12-06:
Assuming this was "good enough" for now. Please re-open this issue if this didn't solve your problem.

closed 2020-12-06 by JnyJny