JnyJny/busylight #14: Multiple of the same Device

cefoster0, 2020-12-03:
Hello! 

I'm using busylight with two Embrava Blynclight(s) connected to a Debian linux system. The multiple lights code appears to be struggling handling two of the same light. 

root@LinuxMini:/dev/usb# busylight list --long
ID: Device Name
 0: Blynclight Standard
	 0: path: b'0003:0004:00'
	 0: vendor_id: 0x2c0d
	 0: product_id: 0x0c
	 0: serial_number: 
	 0: release_number: 0x01
	 0: manufacturer_string: 
	 0: product_string: Blynclight Standard
	 0: usage_page: 0x00
	 0: usage: 0x00
	 0: interface_number: 0x00
 1: Blynclight Standard
	 1: path: b'0004:0004:00'
	 1: vendor_id: 0x2c0d
	 1: product_id: 0x0c
	 1: serial_number: 
	 1: release_number: 0x01
	 1: manufacturer_string: 
	 1: product_string: Blynclight Standard
	 1: usage_page: 0x00
	 1: usage: 0x00
	 1: interface_number: 0x00

but when i try to address and turn on the second one, i get this error

root@LinuxMini:/dev/usb# busylight -l 1 on
Light '1' not in the range of 0..0

JnyJny, 2020-12-03:
I knew this day would come. This a known (to me) limitation of the implementation. In short, the lights are identified by a combination of their vendor identifier and product identifier (0x2C0D:0x000c) which is used to "open" the HID device. Since they both have the "same" Id, the open fails for the second light.  It might be more complicated than that, but that's the gist of the problem. I've ordered a second (hopefully identical) light to test with, it will be a couple of days and a fix shortly afterwards.

JnyJny, 2020-12-04:
Ok, the fix turned out to be less obnoxious than I thought but I haven't been able to test it (Amazon next minute delivery is not yet available in my neighborhood).  If you are feeling adventurous, you can try the new bits and give me a report back?

```console
$ python3 -m pip install -U git+https://github.com/JnyJny/busylight.git
```

cefoster0, 2020-12-04:
thank you so much for the quick response! 

i got this error during the build:
  Cloning https://github.com/JnyJny/busylight.git to /tmp/pip-req-build-km_z91fp
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.7/tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-req-build-km_z91fp/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-km_z91fp/

JnyJny, 2020-12-04:
I'm guessing here, but I think your pip needs an update:

```
$ python3 -m pip install -U pip

```

And then install busylight from the git URL.

cefoster0, 2020-12-04:
it works! great job! 

i had to uninstall the main install before the git one would work, but after upgrading pip, it built successfully and i can target each light independently and all at once!

JnyJny, 2020-12-06:
Good news! Thanks for the bug report and the confirmation that that the fix works.

closed 2020-12-06 by JnyJny