JnyJny/busylight #26: Could you provide a meta package?

redaxmedia, 2021-03-14:
Hello Erik,

it would be awesome to provide a meta package that only contains the USB handling / Light Manager without the Web,- and Command Line interface.

I plan to integrate this lib per device type and don't need anything else:

```
from busylight.lights.embrava import Blynclight

lights = Blynclight.all_lights()

for light in lights:
	light.on((255, 255, 255))
```

JnyJny, 2021-03-14:
It would indeed be awesome :)

Installing busylight into a new virtual environment, `python3 -m pip list` shows:

```
$ python3 -m venv .venv; source .venv/bin/activate
$ python3 -m pip install busylight-for-humans
...
$ python3 -m pip list
Package              Version
-------------------- -------
bitvector-for-humans 0.14.1
busylight-for-humans 0.8.2
click                7.1.2
hidapi               0.10.1
pip                  21.0.1
setuptools           49.2.1
typer                0.3.2
webcolors            1.11.1
```

After installing the webapi optional dependencies , I see:
```
$ python3 -m pip install busylight-for-humans[webapi]
...
$ python3 -m pip list
Package              Version
-------------------- -------
bitvector-for-humans 0.14.1
busylight-for-humans 0.8.2
click                7.1.2
fastapi              0.63.0
h11                  0.12.0
hidapi               0.10.1
pip                  21.0.1
pydantic             1.8.1
setuptools           49.2.1
starlette            0.13.6
typer                0.3.2
typing-extensions    3.7.4.3
uvicorn              0.13.4
webcolors            1.11.1
```

Could you be more specific about your requirements if that doesn't suite your use case?

redaxmedia, 2021-03-14:
Ahh, sorry - I didn't know that the webapi dependencies are optional and won't be installed using:

```
install_requires =
[
	'busylight-for-humans==0.8.2',
]
```

Thanks for letting me know...

closed 2021-03-14 by redaxmedia