JnyJny/busylight #27: Support for Sennheiser Busylight

redaxmedia, 2021-03-14:
Sennheiser has another light that could be added in this library.

https://www.youtube.com/watch?v=B7D7DkVItKI

JnyJny, 2021-03-14:
There is a surprising lack of information about this piece of hardware. Guess I'll have to order one and snoop it to find out how it works.

redaxmedia, 2021-03-19:
@JnyJny Get in touch with the support - I made good experiences asking Luxaflor and Kuando for free examples.

JnyJny, 2021-03-20:
It would seem this light has been discontinued by the manufacturer. I've found it for sale in Europe and Australia, shipping to the US would have doubled the price of the light. Unless someone loans me one it's unlikely this will ever get supported.

closed 2021-03-20 by JnyJny