JnyJny/busylight #28: Plantronics Status Indicator

Hamzouze, 2021-03-25:
Hi Erik,
I'm sorry but I'm not sure this is the right way to ask, but I didn't found any other way to do it.
I'm new on github and by the way I'm not a developper.
Should your program work with the Plantronics Status Indicator device ? It appears it is an Embrava Blynclight OEM.
Trying to run it, no light is recognized, wether it works with the Embrava Connect software.
Nevertheless, I cannot run the Linux post-install activities, maybe this is where my issue stands. (MacOS Catalina 10.15.7).
Thank you for any help that you could provide, if you already know the solution.
KR
Hamza

redaxmedia, 2021-03-25:
I am pretty sure it has another Vendor ID (VID) and Product ID (PID) ... there is a good change you tell Erik these and he can add support for them...

Install:

```
brew install lsusb
```

Run command:

```
lsusb
```

And paste the output here...

Hamzouze, 2021-03-25:
Thank you Henry for your answer and indeed you should be right.

Here is the result of the lsusb command :

> Bus 020 Device 048: ID 047f:d005 Plantronics, Inc. Plantronics Status Indicator

redaxmedia, 2021-03-25:
No problem, the current vendor and product ids are on hexadecimal and not in the format 0000:0000 like I would expect them... let's see what Erik has to say about it.

https://github.com/JnyJny/busylight/blob/master/busylight/lights/embrava/blynclight.py#L15-L16

JnyJny, 2021-03-25:
> Hi Erik,
> I'm sorry but I'm not sure this is the right way to ask, but I didn't found any other way to do it.
> I'm new on github and by the way I'm not a developper.
> Should your program work with the [...]

Hi Hamza, this is exactly the right way to ask. I'm very excited that this is your first GitHub issue and glad you opened it.

First, I don't know for sure that the Plantronics light is an Embrava light for sure, but I've added the Plantronics vendor identifier you provided ( 0x047f ) to the code that supports Embrava devices. With any luck, your light should just work. If not, I'll need to do some trouble shooting. The fix is in busylight-for-humans version 0.8.3 which you can install from a macOS terminal window using:

```console
$ python3 -m pip install -U busylight-for-humans
...
$ busylight on
```

As an aside, if you aren't using the light on a Linux based system, you don't need to do the Linux post-install activities. 

If that doesn't work for you, the output of `busylight list -l` would be very helpful.

Hamzouze, 2021-03-25:
Oh ! Incredible. It works !
Thanks a lot for this enhancement, I didn't imagine you would be able to react that fast !
Now I'm going to be able to sync my sotfphone (Fuze) status 👍 

Thanks a lot Erik and Henry.

JnyJny, 2021-03-25:
Thanks for the quick update on the success! I'm going to close this issue, if you run into problems with the light in the future  please feel free to open a new issue.

closed 2021-03-25 by JnyJny

redaxmedia, 2021-03-25:
@JnyJny Please consider to add Plantronics to the supported vendors inside the README.md aswell.