JnyJny/busylight #40: Wrong naming Agile Innovative

redaxmedia, 2021-05-15:
Hello,

for some reason you used "Innovations" over "Innovative" and therefore the code and docs have incorrect naming.

JnyJny, 2021-05-15:
Passive aggressive issues opened over spelling errors (whose location was not identified) would be better served with pull requests correcting the perceived problems.

closed 2021-05-15 by JnyJny

redaxmedia, 2021-05-15:
Erik, I think you got me wrong here. This issue wasn't meant to be "passive aggressive" ... I just wanted to correct the namespace of the classes in the right way - better to be fixed now than later.

That being said: a pull request has been send and I hope you won't act that offended in the future as I like your library and just want to contribute to it.

JnyJny, 2021-05-15:
Please consider reading this: https://snarky.ca/the-social-contract-of-open-source/

redaxmedia, 2021-05-15:
This is a very negative statement about OpenSource... I suggest you get in touch with me on Discord, Telegram or whatever and we talk while enjoying a beer