JnyJny/busylight #44: Kuando BusyLight does not release usb hub

redaxmedia, 2021-05-15:
Hey,

sorry but there is one issue after another as I implemented your lib to Chroma Feedback today. It seems that Kuando BusyLight has an issue with releasing the usb hub again... for my application I need this to catch exceptions on startup.

https://github.com/redaxmedia/chroma-feedback/blob/master/chroma_feedback/consumer/kuando_busylight/api.py#L24

```
#!/usr/bin/env python3

from time import sleep

from busylight.lights.kuando import BusyLight as API
#from busylight.lights.thingm import Blink1 as API
#from busylight.lights.agile_innovations import BlinkStick as API

first_light = API.first_light()
first_light.on((0, 255, 255))
print(first_light.name)
sleep(1)
first_light.release()
sleep(1)

lights = API.all_lights()
print(lights)

for light in lights:
	print(light.name)
	light.on((255, 0, 0))
	sleep(1)
	light.off()
```

JnyJny, 2021-05-16:
That is an unfortunate side-effect of [how the Kuando Busylight is implemented](https://github.com/JnyJny/busylight/blob/master/docs/devices/kuando.md). If the device doesn't receive a keep-alive write within 30 seconds of the previous write, it will turn itself off. Contrast that with the Embrava Blynclight which will continue whatever it was previously commanded until interrupted, unplugged from the host or the machine is powered off.

I wanted all the lights to operate in a similar manner: if you turn them on, they stay on until turned off. Therefore, the Kuando Busylight (and others) have threads which run the _animation_ indefinitely. This indefinite animation also holds the device open until the current process exits, which means the status of the light cannot be modified from another process.

It turns out, I anticipated this problem when I wrote `busylight.manger.LightManager`. I suggest accessing the lights via the `LightManager` instead of instantiating them directly.

redaxmedia, 2021-05-16:
Thanks for the explanation, I studied the code and figured out the keepalive thread. This was the reason I used the release() method as it should cancel the thread and therefore release the device again!?

As mentioned, I use this just to approve connected devices - is there something we can do to check this another way or properly introduce another helper function for this usecase that does not open a thread?

```python
try
	api.first_light().release()
except (USBLightIOError, USBLightNotFound):
	exit(wording.get('connection_no').format('KUANDO BUSYLIGHT') + wording.get('exclamation_mark'))
```

What is different using the `LightManager` ? I did not have the time to give it a try but should this work with my hacky solution?

Thanks so far

JnyJny, 2021-05-18:
I ran your scripts since what you were reporting shouldn't have occurred and sure enough, there is (was) a bug in the threading code that runs the keep-alive thread which affects only the Kuando BusyLight.  Version 0.12.2 ( d2289fa3e719e01b47e584f51787acdc29e5dfd5) fixes the threading bug and addresses breakage in the web API.  Calling methods on Kunado BusyLight instantiations should no longer block indefinitely.

redaxmedia, 2021-05-18:
Thanks for the new release - it works way better but still has a tiny bug.

Just in case the device has not been read and the thread properly not been opened, the release() method goes into an infinite loop:

```python
#!/usr/bin/env python3

from busylight.lights.kuando import BusyLight as API

first_light = API.first_light()

# uncomment one of the next lines and it works
#first_light.off()
#print(first_light.name)
first_light.release()

lights = API.all_lights()
print(lights)

for light in lights:
	print(light.name)
	light.on((255, 0, 0))
```

redaxmedia, 2021-05-18:
Interesting, can you bring this conclusion into a new release that makes the call `API.first_light().release()` work?

Why not make the timeout adjustable for the user? I have no API in mind but that could make my life properly easier.

JnyJny, 2021-05-18:
I've just pushed (ca44b2f1e368fcd94a20fa8426e82ae92096ee55). Now the test code from the previous comment reports:
```
python3 ./tst.py
first_light dt: 0.01 seconds
         on dt: 0.00 seconds
    release dt: 0.10 seconds
    aqcuire dt: 0.00 seconds
        off dt: 0.00 seconds
```

At the expense of more busy-waiting, the latency of the `release` method has been reduced to 0.1 seconds.

redaxmedia, 2021-05-19:
It works perfect now and is super responsive... thank you so much for that fix and that insane release cycles.

closed 2021-05-19 by redaxmedia