JnyJny/busylight #45: Add udev rule behind the scenes

redaxmedia, 2021-05-16:
This is more of a discussion I like to start, the idea is to make user's live easier by adding the udev rule during install / removing the udev rule during uninstall.

So in case we detect linux as a platform and the user is root, this could be triggered:

## Setup
```
sudo busylight setup
```

that does following behind the scenes:

```
$ busylight udev-rules | tee /etc/udev/rules.d/99-busylight.rules
$ udevadm control --reload
$ udevadm trigger
```

## Teardown

```
sudo busylight teardown:
```

that does following behind the scenes:

```
$ rm /etc/udev/rules.d/99-busylight.rules
$ udevadm control --reload
$ udevadm trigger
```

If the installation is triggered without being a root, there should be the information to run `sudo busylight setup`.

JnyJny, 2021-05-16:
This is a logical step in the evolution of the `udev` sub-command, however what you propose requires the user to trust `busylight` too much.  Yes, they could read the code and determine that there is no threat in escalating `busylight`'s credentials, however most people do not or cannot make that determination for whatever reason. If at some point a bad actor inserts code which exploits that trust, I would be to blame for encouraging this pattern of behavior. I will not allow `busylight` to be a risk for root privilege escalation.

redaxmedia, 2021-05-16:
This are valid points... is there at least something we can do with the suggested exceptions to be thrown?

JnyJny, 2021-05-16:
How do you tell the difference between a light that isn't plugged in versus one that is plugged in but missing udev rules?

redaxmedia, 2021-05-17:
Well, here we have to think outside the box and properly use a `non-udev` approach and create a diff to the `udev` list?

I found this so far: https://stackoverflow.com/questions/8110310/simple-way-to-query-connected-usb-devices-info-in-python

JnyJny, 2021-05-17:
What exactly is it you are asking for? I've already said it's not a good idea to escalate permission in a python program just to automate a simple one-time operation.

Your last question about "suggested exceptions to be thrown" is confusing but I took that to mean issue a diagnostic if no lights are found to alert the user to configure the `udev` subsystem. If `busylight` were a Linux-specific application, I would have no problem with this. However it endeavors to be multi-platform and issuing a diagnostic for Linux would confuse a Mac/BSD/Windows user. To prevent that confusion, I would now have to detect OS versions and issue different diagnostics based on the OS. At this point, `busylight` has diverged from being a "make the USB light turn on and off" tool to a "tool for detecting the OS and warning the user about vagaries of each in the event that USB devices that we want to turn on and off are not available and perhaps change parts of the OS to facilitate maybe making them available and then turn the light on and off" tool. This is an example of "feature creep" and ain't nobody got time for that.

redaxmedia, 2021-05-17:
All right, so let me close this issue then... you are right on most of the points, it would make sense to have this inside another library but not in this one.

closed 2021-05-17 by redaxmedia