JnyJny/busylight #54: Busylight goes off after 25 sec. with CLI on Linux

jchmielniak, 2021-06-11:
Hi,

I can control busylight (Kuando) after installation on Fedora 34 works. After setting it ON (busylight on green) goes the light off after 25 sec. I couldn't'd test it with Python API

JnyJny, 2021-06-11:
Thanks for letting me know. Recent work on the threading model of the lights re-introduced this behavior. If the Kuando Busylight isn't constantly kept "alive" through software intervention, it turns itself off. It's a [weird design decision](https://github.com/JnyJny/busylight/blob/master/docs/devices/kuando.md) but it is what it is. 

I've committed 118a68044cd43214fbd690264ad3dd2db17abb76 which should fix the issue when called from the command line interface. The `busylight` command will not return now, so you will need to put it in the background or consider using the web API to control the light.

```console
$ python3 -m pip install -U busylight-for-humans[webapi]
...
$ busylight serve >& busylight.logs &
$ curl -s http://0.0.0.0:8888/light/0/on
```

JnyJny, 2021-06-11:
FYI, the fix is available in version 0.12.7, just published to pypi.

dynek, 2022-02-09:
Hello @JnyJny - I'm having this specific issue with 0.15.0.

Both with command line `busylight on red` and web server `curl -s http://0.0.0.0:8888/light/0/on/red`.

Any regression ?

JnyJny, 2022-02-09:
I haven't seen a regression, however my testing resources are limited:
- macOS 12.2, python 3.8, python 3.9 and python 3.10
- RaspberryPi 3 running a Debian-alike using python 3.8, python 3.8 and python 3.10

What operating system and version of python are using where you see the failure?

Can you post up the debug output from the command `busylight -D on red` ?

There should be entries in the output similar to:
```
2022-02-09 09:45:33.649 | INFO     | busylight.lights.kuando.busylight:keepalive:100 - keepalive sent
2022-02-09 09:45:33.650 | DEBUG    | busylight.lights.usblight:update:538 - Busylight update for b'\x8f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\xff\x03\x8c'
```

dynek, 2022-02-09:
macOS 12.1 & Python 3.9.10 (homebrew).

Here's the output:
```
2022-02-09 16:57:01.493 | INFO     | busylight.manager:update:172 - Manager has 0 lights
2022-02-09 16:57:01.493 | DEBUG    | busylight.lights.usblight:all_lights:173 - USBLight querying subclass lights...
2022-02-09 16:57:01.493 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: BlinkStick
2022-02-09 16:57:01.493 | DEBUG    | busylight.lights.usblight:first_light:108 - BlinkStick looking for first unclaimed light...
2022-02-09 16:57:01.502 | DEBUG    | busylight.lights.usblight:first_light:123 - BlinkStick found no unclaimed lights.
2022-02-09 16:57:01.503 | DEBUG    | busylight.lights.usblight:all_lights:170 - BlinkStick found 0 lights
2022-02-09 16:57:01.503 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Blynclight
2022-02-09 16:57:01.503 | DEBUG    | busylight.lights.usblight:first_light:108 - Blynclight looking for first unclaimed light...
2022-02-09 16:57:01.505 | DEBUG    | busylight.lights.usblight:first_light:123 - Blynclight found no unclaimed lights.
2022-02-09 16:57:01.505 | DEBUG    | busylight.lights.usblight:all_lights:170 - Blynclight found 0 lights
2022-02-09 16:57:01.505 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Busylight
2022-02-09 16:57:01.506 | DEBUG    | busylight.lights.usblight:first_light:108 - Busylight looking for first unclaimed light...
2022-02-09 16:57:01.508 | DEBUG    | busylight.lights.usblight:first_light:123 - Busylight found no unclaimed lights.
2022-02-09 16:57:01.508 | DEBUG    | busylight.lights.usblight:all_lights:170 - Busylight found 0 lights
2022-02-09 16:57:01.508 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Flag
2022-02-09 16:57:01.508 | DEBUG    | busylight.lights.usblight:first_light:108 - Flag looking for first unclaimed light...
2022-02-09 16:57:01.509 | DEBUG    | busylight.lights.usblight:first_light:111 - Flag entry found for 1240
2022-02-09 16:57:01.509 | DEBUG    | busylight.lights.usblight:first_light:113 - entry: {'path': b'DevSrvsID:4295159928', 'vendor_id': 1240, 'product_id': 63560, 'serial_number': '', 'release_number': 2, 'manufacturer_string': 'BLL Company ApS', 'product_string': 'BLL Lamp', 'usage_page': 65280, 'usage': 1, 'interface_number': 0}
2022-02-09 16:57:01.509 | DEBUG    | busylight.lights.usblight:__init__:231 - Flag init 4d8 f848 b'DevSrvsID:4295159928' False
2022-02-09 16:57:01.509 | DEBUG    | busylight.lights.usblight:acquire:457 - Flag.acquire(reset=False)
2022-02-09 16:57:01.510 | DEBUG    | busylight.lights.usblight:acquire:459 - lock acquired
2022-02-09 16:57:01.510 | SUCCESS  | busylight.lights.usblight:acquire:475 - Flag hid.open_path('DevSrvsID:4295159928') was a success None
2022-02-09 16:57:01.512 | DEBUG    | busylight.lights.usblight:all_lights:167 - Flag added a light BLL Lamp: 0x04d8:0xf848
2022-02-09 16:57:01.512 | DEBUG    | busylight.lights.usblight:first_light:108 - Flag looking for first unclaimed light...
2022-02-09 16:57:01.514 | DEBUG    | busylight.lights.usblight:first_light:111 - Flag entry found for 1240
2022-02-09 16:57:01.514 | DEBUG    | busylight.lights.usblight:first_light:113 - entry: {'path': b'DevSrvsID:4295159928', 'vendor_id': 1240, 'product_id': 63560, 'serial_number': '', 'release_number': 2, 'manufacturer_string': 'BLL Company ApS', 'product_string': 'BLL Lamp', 'usage_page': 65280, 'usage': 1, 'interface_number': 0}
2022-02-09 16:57:01.514 | DEBUG    | busylight.lights.usblight:__init__:231 - Flag init 4d8 f848 b'DevSrvsID:4295159928' False
2022-02-09 16:57:01.514 | DEBUG    | busylight.lights.usblight:acquire:457 - Flag.acquire(reset=False)
2022-02-09 16:57:01.514 | DEBUG    | busylight.lights.usblight:acquire:459 - lock acquired
2022-02-09 16:57:01.515 | ERROR    | busylight.lights.usblight:acquire:463 - hid_open open failed for open_path('DevSrvsID:4295159928')
2022-02-09 16:57:01.515 | ERROR    | busylight.lights.usblight:first_light:120 - Flag (1240, 63560, 'DevSrvsID:4295159928') for 1240
2022-02-09 16:57:01.515 | DEBUG    | busylight.lights.usblight:first_light:123 - Flag found no unclaimed lights.
2022-02-09 16:57:01.515 | DEBUG    | busylight.lights.usblight:all_lights:170 - Flag found 1 lights
2022-02-09 16:57:01.515 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Status_Indicator
2022-02-09 16:57:01.515 | DEBUG    | busylight.lights.usblight:first_light:108 - Status_Indicator looking for first unclaimed light...
2022-02-09 16:57:01.517 | DEBUG    | busylight.lights.usblight:first_light:123 - Status_Indicator found no unclaimed lights.
2022-02-09 16:57:01.517 | DEBUG    | busylight.lights.usblight:all_lights:170 - Status_Indicator found 0 lights
2022-02-09 16:57:01.517 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Blink1
2022-02-09 16:57:01.517 | DEBUG    | busylight.lights.usblight:first_light:108 - Blink1 looking for first unclaimed light...
2022-02-09 16:57:01.518 | DEBUG    | busylight.lights.usblight:first_light:123 - Blink1 found no unclaimed lights.
2022-02-09 16:57:01.518 | DEBUG    | busylight.lights.usblight:all_lights:170 - Blink1 found 0 lights
2022-02-09 16:57:01.518 | DEBUG    | busylight.lights.usblight:all_lights:176 - USBLight found 1 in total.
2022-02-09 16:57:01.518 | INFO     | busylight.manager:update:175 - After all_lights, Manager has 1 lights
2022-02-09 16:57:01.520 | INFO     | busylight.manager:update:182 - Manager has 1 lights after update
2022-02-09 16:57:01.520 | DEBUG    | busylight.lights.usblight:update:538 - Flag update for b'\x01\xff\x00\x00\x00\x00\x00\x00'
2022-02-09 16:57:01.525 | DEBUG    | busylight.lights.usblight:update:538 - Flag update for b'\x01\xff\xff\x00\x00\x00\x00\x00'
```

JnyJny, 2022-02-09:
Thanks for the debug output! Also... _weird_. Are there any physical markings on the light to identify? I'm asking because I *think* this is a very old Kuando UC Omega that was probably originally manufactured by Luxafor (or a subcontractor or something like that). The Vendor ID for this light is 0x04d8 which the `Flag` module claims, and that module definitely doesn't know how to talk to a Kuando Busylight and keep it illuminated.  I can add some product_id filtering to accept/reject lights which should fix the problem. Those Kuando products have always been problem children (not your fault).

dynek, 2022-02-09:
Thanks for the info. Can't really say how old it is. I found it lying around at work and thought about using it.
It just says "Kuando (tm) busy light". It looks like this:
[image]

For the moment, I worked around the problem with a while loop and a temp file containing the color (triggered via an Alfred Workflow).

No need to invest more time into this on your side, except if you feel it could make sense.

JnyJny, 2022-02-09:
Thanks for the photo, it's a UC Busylight Alpha according to the internets, and I don't have one of those in my testing stable.  Still weird that the `Flag` module is operating the light enough to turn it on.

dynek, 2022-02-10:
Perfect @JnyJny ! Thanks a bunch, it's working like a charm now, no more while loop ;-)