JnyJny/busylight #55: Issue trying to install

alexander-paterson, 2021-06-11:
I'm trying to install this on my Ubuntu machine.

I've ran the command `python3 -m pip install busylight-for-humans` and it seemed to do its thing.

Now as I'm using Linux I'm trying to run the commands to access the USB devices.

When I run `busylight udev-rules -o 99-busylights.rules` I get the message `busylight: command not found` and I'm not sure what I might have done wrong.

JnyJny, 2021-06-11:
If you installed it as a non-root user, likely the script was installed into a directory that is not in your  `$PATH` environment variable.

I don't have an Ubuntu machine to verify, but with Debian and CentOS `python3 -m pip install` puts script stubs in `$HOME/.local/bin`.  If that path is missing from `$PATH` then your shell can't find the executable.

```
$ echo $PATH 
...
```
That's my best guess.

alexander-paterson, 2021-06-11:
Thanks, that directory wasn't in my path. Adding it has gotten the script working.

closed 2021-06-11 by alexander-paterson