JnyJny/busylight #56: Installation on Windows with Kuando

Klauz123, 2021-06-12:
Hi there,

i am new to python and i am having a tough time working with the kuando light on windows. On mac everything was fine.
I tried to test with following code:
`from busylight.lights.kuando import BusyLight`
`light = BusyLight.first_light()`
`light.on((255, 0, 0))`
I get the following result:
`Traceback (most recent call last):
  File "C:\Users\MyPC\Desktop\Python\test.py", line 4, in <module>
    light.on((255, 0, 0))
  File "C:\Users\MyPC\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\busylight\lights\kuando\busylight.py", line 49, in on
    self.state.line0 = instruction.value
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\contextlib.py", line 124, in __exit__
    next(self.gen)
  File "C:\Users\MyPC\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\busylight\lights\usblight.py", line 412, in batch_update
    self.update()
  File "C:\Users\MyPC\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\busylight\lights\usblight.py", line 507, in update
    raise USBLightIOError(f"write returned {nbytes}")
busylight.lights.exceptions.USBLightIOError: write returned 65`

Does anyone have any idea what I am doing wrong?``

JnyJny, 2021-06-12:
As you might guess, I develop on Mac and test on Linux but have never actually tested on Windows. The Kuando Busylight control packet is 64 bytes long and 65 bytes were written. This could be a one-off bug in my code or some sort of Windows-specific interaction that I hadn't anticipated. I don't think this is a problem caused by you doing something wrong.  I'll investigate and see what I can figured out.

JnyJny, 2021-06-12:
If you could describe to me how you installed busylight-for-humans on Windows that would be an enormous help to me.

Klauz123, 2021-06-12:
Ofcourse!
I had almost a fresh installation of Windows 10. I installed python via the windows store and installed the busylight-for-humans package via 
`python -m pip install busylight-for-humans`

after that i created a test.py file with the above code.

JnyJny, 2021-06-12:
Awesome. I have been able to replicate the error you reported. I *suspect* that it's because `[libusb](https://github.com/libusb/libusb)` isn't installed ( it's not installed on my Windows either ). Investigating further. If anything, `busylight` should have failed sooner than that with a better message.

JnyJny, 2021-06-12:
Poking around some more, missing libusb doesn't seem to be the problem. Which is great because debugging this is frustrating enough :) It looks like Windows may be a little more picky about the control packet written to the device, but may be an easy fix. Stand by.

JnyJny, 2021-06-12:
And no, I was not so lucky. It appears that the Kuando Busylight `keepalive_thread` is dying and restarting in an epically bad way. Possibly a permissions problem, possibly a bad assumption with respect to threads, or something else. Could you reply back with the output of:
```console
$ busylight -D list
```
You'll probably need to kill the script, dependent on what Windows shell technology you're using (Control-C in `cmd` worked for me).

Klauz123, 2021-06-12:
Uhm, could you explain to me how I can set up the cmd.exe commands in windows for busylight? I haven't figured it out... If i type busylight the response show that the command is unknown.

Like i said - I'm new :-D

JnyJny, 2021-06-12:
Likely your PATH variable doesn't include the filesystem path to your python executable. I was bit by exactly the same thing which led to a frustrating several hours of futzing with Windows.  My solution to the problem was:

1. In Settings, uninstall the Python package from the Microsoft Store
2. Go to https://python.org, hover over the `Download` tab and click the **Download** button. 
3. After it downloads, click "Open file" in the Downloads drop down menu. 
4. Install Python and click the option to add the python executable to the system PATH environment variable.

After I did this, I was able to start a `cmd` terminal and run python:
```console
C:\Users\ejo\local\busylight> python
Python 3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

After getting python to run from a prompt, you will likely need to re-install `busylight`:

```console
C:\Users\ejo\local\busylight> python -m pip install busylight-for-humans
...
```

Assuming that completes successfully, you can then run the `busylight` command-line tool using:

```console
C:\Users\ejo\local\busylight> busylight --help
...
```

I was hoping to see if your debug output matches mine for:
```console
c:\Users\ejo\local\busylight> busylight -D list
...
```

JnyJny, 2021-06-13:
So I’ve root caused the problem; I assumed that a HID USB device could only be opened successfully one time (exclusive open). This is true for macOS and Linux but not true for Windows. The entire light discovery mechanism depends on this assumption unfortunately. I’ll need to re-think how I manage devices to make things work on Windows. Sorry if this doesn’t make sense, I’m hoping to have a solution for this pretty quick now that I understand the problem.

Klauz123, 2021-06-13:
You are awesome! Thank you!

redaxmedia, 2021-06-27:
Hey Erik, I experienced the same problem with all busylight powered devices in my project. Even a listing does not seem to be possible...

JnyJny, 2021-06-29:
Fixing this problem on Windows will require some re-architecting of the USBLight class which I do not have time for right now, but will do at some point in the future. In the meantime, the README will be updated to indicate that BusyLight does not work on Windows right now.

redaxmedia, 2021-08-10:
@JnyJny Let me know if you need help or a sparring partner to get this issue solved.

JnyJny, 2025-03-05:
An additional fix for Windows 11 documented in #434 has been integrated which should allow use of lights on Windows 10 a Windows 11. Crosses fingers.

closed 2025-03-05 by JnyJny