JnyJny/busylight #74: Add additional Embrava vendor ID

hewasajumperboy, 2021-09-09:
Hey,

I'm trying to use my (probably older) light with this project.  It works in the previous, depreciated blynclight repo and looking at embrava/blynclight.py the vendor id for my light doesn't show up in the list.  Output from my pi below:

````
pi@nightlight:~ $ busylight on
Light '0' not in the range of 0..-1
pi@nightlight:~ $ dmesg | tail
[ 7218.521658] Voltage normalised (0x00000000)
[ 9518.695285] usb 1-1.2: USB disconnect, device number 4
[17252.456676] usb 1-1.3: USB disconnect, device number 6
[17254.285218] usb 1-1.3: new low-speed USB device number 7 using dwc_otg
[17254.429614] usb 1-1.3: New USB device found, idVendor=0e53, idProduct=2516, bcdDevice= 1.00
[17254.429637] usb 1-1.3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[17254.429652] usb 1-1.3: Product: Blynclight
[17254.436605] hid-generic 0003:0E53:2516.000A: hiddev96,hidraw0: USB HID v1.10 Device [Blynclight] on usb-3f980000.usb-1.3/input0
````

JnyJny, 2021-09-09:
TIL: you can transfer issues between repos!

Transferred this issue from the blynclight repo to this repo, just because I've never had to do that before.

JnyJny, 2021-09-09:
Fixed with commit a468d6a84f19261fc4d656d0082d852fc3ab090b and should be available on pypi with version 0.12.10:

```console
$ python3 -m pip install -U busylight-for-humans
```

hewasajumperboy, 2021-09-10:
Awesome!  Looks like this change fixed it, I haven't rebooted yet but it was able to control the light as sudo.

JnyJny, 2021-09-13:
You shouldn't need to reboot to use the light, look at the `udev-rules` sub-command of the `busylight` CLI. There is more info in the README.md that talks about how to enable USB devices for non-root users when using Linux.  I'm going to close this issue since the light is now recognized.

closed 2021-09-13 by JnyJny

hewasajumperboy, 2021-09-13:
Sounds good.  I've gone through the udev-rules post install steps, I had to append the new device id to the *.rules file but no dice after reloading and/or rebooting.  Running as a regular user gives the "Light 0 not in range..." message.  Running as sudo works to control the light.

I haven't looked into it more, it could be some weirdness with running raspbian and once I have some more time I was going to test it with ubuntu server and see if it had the same behavior.