JnyJny/busylight #82: [BUG] Luxafor flag - cannot use blink and would there be other modes?

magic-muschrum, 2021-11-07:
**Software Versions:**
 - Operating System: macOS 11.6 on Apple M1
 - Python version: 3.8.2
 - BusyLight version: 0.13.1

**General Type of Problem**
- [ ] Installation
- [x] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [ ] other

**Describe the Problem**
Connected a Luxafor flag to the MacBook Pro, gets recognized just fine, busylight on and off work fine as well, but busylight blink does nothing or would switch the light off. Have the debug output in the attached file but could not see the problem with it. Is blink not supported for the flag? 

Also tried fade, strobe, wave and others but was not sure, if they are really supported, as they are not a command. So I guess they would only serve in my own python application as described on the code examples?

Maybe it is just me and I use the command wrong or did sth else incorrectly. I also assumed that I do not have to touch any udev rules on MacOS as it was suggested for Linux?!

Thank you for your help and this great tool in general 😄 

**Expected Behavior**
Flag to blink.

**Error Output**
No error, just new command line with cursor. Debugger output attached in file
[Debug output luxafor flag.pdf](https://github.com/JnyJny/busylight/files/7492929/Debug.output.luxafor.flag.pdf)

JnyJny, 2021-11-08:
Yup, that's a real bug. Seeing the same behavior with my Luxafor Flag. Doesn't look like the state of the light isn't being updated correctly, it will take some time to dig into this.

JnyJny, 2021-11-09:
Looks like I broke or forgot to completely implement the Luxafor Flag commands. The `FlagStrobeCommand` and `FlagWaveCommand` classes were missing an implementation of the `color` property.  I probably should have a `FlagBaseCommand` class that implements the color property, but I'll do that later.  This fix is 168161f20dd1651cf00b2dbfd6aacddb43377035 and is tagged as version 0.13.2.

The `busylight` command only exposes capability that is common across all supported lights; `on`, `off` and `blink`.  If you want to access other capabilities you'll need to write a little bit of code.

magic-muschrum, 2021-11-09:
Hello, thank you for this quick response. Yes, blink works, now. Thank you. For the rest, I will try to find the time and write some code :) Looking forward to learn sth new.

I just have a question concerning the blinking speed. Besides that it is a bit hard to see the difference between slow, medium, and fast, I also think that the different colors blink with different speeds even when the same level is set. Could this be or is this just my eyes, playing a trick on me?

Thank you!

JnyJny, 2021-11-09:
Glad to hear that blink is working again for you.

WRT to blink speeds; I just set the bits and what the hardware decides to do with those bits is out of my hands.  That said, I did not do much tuning to the mapping from "slow", "medium", and "fast" to the 8-bit speed field for the Flag. I'll be happy to accept pull requests that address that mapping.  The code in question is in `busylight.lights.luxafor.flag.blink()`.

closed 2021-11-09 by JnyJny