JnyJny/busylight #84: Auto-starting Busylight Serve API[FEATURE]

adenijisegun, 2021-11-18:
I have found busylight useful in so many use cases and I appreciate the effort put into this starting from Blynclight days.

I wish there is a way to have the WEBAPI uvicorn auto-start, I have tried a plethora of options, but all of them failed.

I tried lxterminal autostart at boot-up; running a service; rclocal; crontab; desktop file.  Issued encountered ranged from permissions to import errors.

The auto-start of the WEBAPI will be useful in many unattended use cases, where a setup like raspberry pi can boot up and have the WEBAPI running and responding to input via a web interface.

JnyJny, 2021-11-19:
This isn't something I would build into `busylight` per-se. But I wouldn't be adverse to collecting various startup / auto-start configuration files in a "contrib" directory in the repo.  For Linux distributions using `systemd` , the right answer is writing a small service definition. MacOS needs a `launch` service configuration and I have no idea what Windows offers.

adenijisegun, 2021-11-19:
Thank you for the insight.
I agree systems will more than do the work on Linux distros, after sorting out permission issues encountered, I have eventually got a service written up and running on a pi ready to be integrated into a bunch of applications.

closed 2021-11-19 by adenijisegun

bbgit, 2022-02-23:
@adenijisegun Would you be willing to share the service definition that you came up with for the pi? I'm trying to do the same and can't seem to get it working. 

Thanks!