JnyJny/busylight #99: Access to light does not persist across restarts / "Light '0' not in the range of 0..-1"[BUG]

DailytheNoob, 2022-01-20:
**Software Versions:**
 - Operating System: Raspberry Pi OS 11
 - Python version: 3.9.2
 - BusyLight version: latest (0.13.2)
 - Luxafor Flag

**General Type of Problem**
- Installation

**Describe the Problem**
After having everything working via API and everything I restarted to move the flag to its final location and it no longer works. I double checked that it was not the cable by trying the one I was testing with with no success. I also do not get the "startup animation" when I plug it in. I am using the cable that came with the flag.
The udev rules are in `/etc/udev/rules.d`, double checked that.
I restarted it with the flag unplugged and the systemd service disabled just in case that was causing any issues with no change in the outcome.

**Expected Behavior**
I expect not to have to set it up every time the RasPi is restarted

**Error Output**
```
$ busylight on
Light '0' not in the range of 0..-1
```

DailytheNoob, 2022-01-20:
jesus christ, it was the cable after all. fuck me

closed 2022-01-20 by DailytheNoob

JnyJny, 2022-01-21:
Whew. Glad you solved the problem!

JnyJny, 2022-01-21:
To be fair, the error message:
```
$ busylight on
Light '0' not in the range of 0..-1
```
is pretty awful. I'll throw in a fix for that for free :)