JnyJny/busylight #110: [BUG] Kuando Busylight UC Alpha not detected

klugerama, 2022-02-15:
**Software Versions:**
 - Operating System: Ubuntu 20.04.03
 - Python version: 3.8
 - BusyLight version: 0.15.1

**General Type of Problem**
- [ ] Installation
- [x] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [ ] other

**Describe the Problem**
Kuando Busylight UC Alpha not detected

**Expected Behavior**
Kuando Busylight UC Alpha... detected

**Error Output**
```
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 018: ID 17e9:436e DisplayLink 
Bus 002 Device 017: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 016: ID 17e9:436e DisplayLink USB3.0 Hub             
Bus 002 Device 015: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 014: ID 2109:0211 VIA Labs, Inc. USB3.0 Hub             
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 029: ID 413c:b080 Dell Computer Corp. 
Bus 001 Device 036: ID 27bb:3bca                            <--- Kuando Busylight UC Alpha
Bus 001 Device 031: ID 413c:3012 Dell Computer Corp. Optical Wheel Mouse
Bus 001 Device 030: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 028: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 027: ID 2109:2211 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 009: ID 8087:0026 Intel Corp. 
Bus 001 Device 006: ID 0c45:6a0c Microdia Integrated_Webcam_HD
Bus 001 Device 004: ID 27c6:533c Shenzhen Goodix Technology Co.,Ltd. FingerPrint
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
$ busylight -D list
2022-02-15 13:41:31.769 | DEBUG    | busylight.lights.usblight:all_lights:173 - USBLight querying subclass lights...
2022-02-15 13:41:31.770 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: BlinkStick
2022-02-15 13:41:31.770 | DEBUG    | busylight.lights.usblight:first_light:118 - BlinkStick looking for first unclaimed light...
2022-02-15 13:41:31.772 | DEBUG    | busylight.lights.usblight:first_light:133 - BlinkStick found no unclaimed lights.
2022-02-15 13:41:31.772 | DEBUG    | busylight.lights.usblight:all_lights:170 - BlinkStick found 0 lights
2022-02-15 13:41:31.772 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Blynclight
2022-02-15 13:41:31.772 | DEBUG    | busylight.lights.usblight:first_light:118 - Blynclight looking for first unclaimed light...
2022-02-15 13:41:31.772 | DEBUG    | busylight.lights.usblight:first_light:133 - Blynclight found no unclaimed lights.
2022-02-15 13:41:31.772 | DEBUG    | busylight.lights.usblight:all_lights:170 - Blynclight found 0 lights
2022-02-15 13:41:31.772 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Busylight
2022-02-15 13:41:31.772 | DEBUG    | busylight.lights.usblight:first_light:118 - Busylight looking for first unclaimed light...
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:first_light:121 - Busylight entry found for 10171
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:first_light:123 - entry: {'path': b'1-3.1.4:1.0', 'vendor_id': 10171, 'product_id': 15306, 'serial_number': '', 'release_number': 256, 'manufacturer_string': 'PLENOM APS', 'product_string': 'BUSYLIGHT', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
2022-02-15 13:41:31.774 | ERROR    | busylight.lights.usblight:first_light:130 - Busylight  for 10171
2022-02-15 13:41:31.774 | ERROR    | busylight.lights.kuando.busylight:release:150 - unsupported format string passed to NoneType.__format__
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:first_light:133 - Busylight found no unclaimed lights.
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:all_lights:170 - Busylight found 0 lights
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Flag
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:first_light:118 - Flag looking for first unclaimed light...
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:first_light:133 - Flag found no unclaimed lights.
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:all_lights:170 - Flag found 0 lights
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Status_Indicator
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:first_light:118 - Status_Indicator looking for first unclaimed light...
2022-02-15 13:41:31.774 | DEBUG    | busylight.lights.usblight:first_light:133 - Status_Indicator found no unclaimed lights.
2022-02-15 13:41:31.775 | DEBUG    | busylight.lights.usblight:all_lights:170 - Status_Indicator found 0 lights
2022-02-15 13:41:31.775 | DEBUG    | busylight.lights.usblight:all_lights:163 - Searching for lights belonging to: Blink1
2022-02-15 13:41:31.775 | DEBUG    | busylight.lights.usblight:first_light:118 - Blink1 looking for first unclaimed light...
2022-02-15 13:41:31.775 | DEBUG    | busylight.lights.usblight:first_light:133 - Blink1 found no unclaimed lights.
2022-02-15 13:41:31.775 | DEBUG    | busylight.lights.usblight:all_lights:170 - Blink1 found 0 lights
2022-02-15 13:41:31.775 | DEBUG    | busylight.lights.usblight:all_lights:176 - USBLight found 0 in total.
No lights found.
```

JnyJny, 2022-02-16:
Thank you for the great bug report!

It looks like there are multiple vendor and product ids associated with the UC Busylight Alpha product which makes things interesting but not impossible. I should have a fix committed within the next day for you to try.

JnyJny, 2022-02-17:
I just committed 5ea8ef919444c7b4bfe61cd1dc29f07a5ec2f618 which should fix the problem you reported. The recently published version 0.15.2 includes this commit. Please let me know if that fixes the problem for you.

klugerama, 2022-02-18:
Awesome, thank you! Unfortunately I can't check for a few days (WFH) but I will be sure to let you know.

redaxmedia, 2022-02-20:
Erik, did you ever consider to contact vendors to ask for a list of product ids?

klugerama, 2022-02-23:
I'm sorry, I'm not smart enough to figure out how to run from source. Is there some documentation about how to do that?

JnyJny, 2022-02-23:
No need to run or install from the git source repository, you can update your current installation using:

```
$ python3 -m pip install -U busylight-for-humans
```

klugerama, 2022-02-23:
Ah fantastic. Confirmed, that worked, TYVM!

closed 2022-02-23 by klugerama