JnyJny/busylight #113: [BUG] Busylight Omega not detected

dfangl, 2022-02-21:
**Software Versions:**
 - Operating System: Arch Linux
 - Python version: 3.8.10
 - BusyLight version: 0.15.2

**General Type of Problem**
- [ ] Installation
- [x] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [ ] other

**Describe the Problem**
Busylight Omega is not detected. It seems a new revision is out, with the product ID `0x3BCF`. Currently, only 0x3BCD is defined.
Also, busylight throws an error for a failing assertion on exit.

**Expected Behavior**
Correct detection of omega light and no assertion error on exit.

**Error Output**
Error missing Product ID:
```
$ busylight -D on red
2022-02-21 19:24:06.382 | INFO     | busylight.manager:update:172 - Manager has 0 lights
2022-02-21 19:24:06.382 | DEBUG    | busylight.lights.usblight:all_lights:177 - USBLight querying subclass lights...
2022-02-21 19:24:06.382 | DEBUG    | busylight.lights.usblight:all_lights:167 - Searching for lights belonging to: BlinkStick
2022-02-21 19:24:06.382 | DEBUG    | busylight.lights.usblight:first_light:118 - BlinkStick looking for first unclaimed light...
2022-02-21 19:24:06.384 | DEBUG    | busylight.lights.usblight:first_light:137 - BlinkStick found no unclaimed lights.
2022-02-21 19:24:06.384 | DEBUG    | busylight.lights.usblight:all_lights:174 - BlinkStick found 0 lights
2022-02-21 19:24:06.384 | DEBUG    | busylight.lights.usblight:all_lights:167 - Searching for lights belonging to: Blynclight
2022-02-21 19:24:06.384 | DEBUG    | busylight.lights.usblight:first_light:118 - Blynclight looking for first unclaimed light...
2022-02-21 19:24:06.384 | DEBUG    | busylight.lights.usblight:first_light:137 - Blynclight found no unclaimed lights.
2022-02-21 19:24:06.385 | DEBUG    | busylight.lights.usblight:all_lights:174 - Blynclight found 0 lights
2022-02-21 19:24:06.385 | DEBUG    | busylight.lights.usblight:all_lights:167 - Searching for lights belonging to: Busylight
2022-02-21 19:24:06.385 | DEBUG    | busylight.lights.usblight:first_light:118 - Busylight looking for first unclaimed light...
2022-02-21 19:24:06.387 | DEBUG    | busylight.lights.usblight:first_light:121 - Busylight entry found for vendor_id 27bb
2022-02-21 19:24:06.387 | DEBUG    | busylight.lights.usblight:first_light:125 - entry: {'path': b'1-1:1.0', 'vendor_id': 10171, 'product_id': 15311, 'serial_number': '922092FF001024FF13FFFFFF8102FFFF9102FFFF', 'release_number': 256, 'manufacturer_string': 'PLENOM A/S', 'product_string': 'BUSYLIGHT OMEGA', 'usage_page': 0, 'usage': 0, 'interface_number': 0}
2022-02-21 19:24:06.387 | ERROR    | busylight.lights.usblight:first_light:132 - Busylight Requested device product_id not supported: product_id=0x3bcf for vendor_id 27bb
2022-02-21 19:24:06.387 | ERROR    | busylight.lights.usblight:identifier:384 - unsupported format string passed to NoneType.__format__
Traceback (most recent call last):

  File "/home/daniel/projects/busylight/.venv/bin/busylight", line 8, in <module>
    sys.exit(cli())
    │   │    └ <typer.main.Typer object at 0x7f6065886df0>
    │   └ <built-in function exit>
    └ <module 'sys' (built-in)>
  File "/home/daniel/projects/busylight/.venv/lib/python3.8/site-packages/typer/main.py", line 214, in __call__
    return get_command(self)(*args, **kwargs)
           │           │      │       └ {}
           │           │      └ ()
           │           └ <typer.main.Typer object at 0x7f6065886df0>
           └ <function get_command at 0x7f6064d2c700>
  File "/home/daniel/projects/busylight/.venv/lib/python3.8/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
           │    │     │       └ {}
           │    │     └ ()
           │    └ <function BaseCommand.main at 0x7f6064d54280>
           └ <TyperGroup main-callback>
  File "/home/daniel/projects/busylight/.venv/lib/python3.8/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
         │    │      └ <click.core.Context object at 0x7f6064c7d3a0>
         │    └ <function MultiCommand.invoke at 0x7f6064d56310>
         └ <TyperGroup main-callback>
  File "/home/daniel/projects/busylight/.venv/lib/python3.8/site-packages/click/core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
           │               │       │       │      └ <click.core.Context object at 0x7f60649dc4f0>
           │               │       │       └ <function Command.invoke at 0x7f6064d54d30>
           │               │       └ <TyperCommand on>
           │               └ <click.core.Context object at 0x7f60649dc4f0>
           └ <function MultiCommand.invoke.<locals>._process_result at 0x7f60649d7670>
[... 118 lines cut ...]
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:all_lights:174 - Flag found 0 lights
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:all_lights:167 - Searching for lights belonging to: Status_Indicator
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:first_light:118 - Status_Indicator looking for first unclaimed light...
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:first_light:137 - Status_Indicator found no unclaimed lights.
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:all_lights:174 - Status_Indicator found 0 lights
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:all_lights:167 - Searching for lights belonging to: Blink1
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:first_light:118 - Blink1 looking for first unclaimed light...
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:first_light:137 - Blink1 found no unclaimed lights.
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:all_lights:174 - Blink1 found 0 lights
2022-02-21 19:25:07.908 | DEBUG    | busylight.lights.usblight:all_lights:180 - USBLight found 1 in total.
2022-02-21 19:25:07.908 | INFO     | busylight.manager:update:175 - After all_lights, Manager has 1 lights
2022-02-21 19:25:07.908 | INFO     | busylight.manager:update:182 - Manager has 1 lights after update
2022-02-21 19:25:07.909 | DEBUG    | busylight.lights.usblight:stop_animation:541 - 'Busylight' object has no attribute '_animation_thread'
2022-02-21 19:25:07.909 | DEBUG    | busylight.lights.usblight:update:554 - Busylight update for b'\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\xff\x03\r'
2022-02-21 19:25:07.909 | DEBUG    | busylight.lights.usblight:stop_animation:541 - 'Busylight' object has no attribute '_animation_thread'
2022-02-21 19:25:07.909 | DEBUG    | busylight.lights.usblight:update:554 - Busylight update for b'\x10\x00d\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\xff\x03q'
^C2022-02-21 19:25:10.286 | DEBUG    | busylight.lights.usblight:stop_animation:541 - 'Busylight' object has no attribute '_animation_thread'
2022-02-21 19:25:10.287 | DEBUG    | busylight.lights.usblight:update:554 - Busylight update for b'\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\xff\x03\r'
python: io.c:2115: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
[1]    215416 IOT instruction (core dumped)  busylight -D on red
```

### Additional Infos
I will file a quick PR for the missing ID myself, not sure what goes on with the wrong assertion though, and cannot test with other devices if this is something new with the new Product ID.
Is this error currently expected?

JnyJny, 2022-02-21:
Hi and thanks for the bug report and the pull request with the new product_id value!

The other thing you noticed is the result of my decision to use threads to update the Busylight family of devices in the background. Unfortunately, HIDAPI doesn't support accessing HID devices from multiple threads and has recently started crashing under MacOS in the tests (and presumably Linux too, I haven't looked to closely at the results of the automated testing). 

I'm in the process of re-writing with an eye towards using asyncio to keep everything in a single thread while taking advantage of an asyncio event loop to deliver keep-alive messages to Busylights. Easily my least favorite light since it needs so much hand-holding. 

No ETA for the re-write as I've just begun this weekend, however it's giving me the opportunity to clean up a bunch of kludges.

dfangl, 2022-02-21:
Thanks for the quick response! If you need any help, or help testing, I would be happy to lend a hand!

closed 2022-03-29 by JnyJny