JnyJny/busylight #124: [BUG] ImportError: cannot import name 'get_terminal_size' from 'click.termui'

jsavin, 2022-03-28:
**Software Versions:**
 - Operating System: macOS 10.15.7
 - Python version: 3.9
 - BusyLight version: 0.14.0

**General Type of Problem**
- [ ] Installation
- [x] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [ ] other

**Describe the Problem**
The `busylight` CLI command always fails with an import error on `get_terminal_size` from `click.termui` because `get_terminal_size` was removed in `click` v8.1.

I was able to work around this issue by downgrading `click` to v8.0 via:
`% pip3 install --upgrade click==8.0`

**Expected Behavior**
Expected `busylight` CLI commands to work "out of the box" without having to downgrade `click`.

**Error Output**

```
Traceback (most recent call last):
  File "/usr/local/bin/busylight", line 5, in <module>
    from busylight.__main__ import cli
  File "/usr/local/lib/python3.9/site-packages/busylight/__main__.py", line 8, in <module>
    import typer
  File "/usr/local/lib/python3.9/site-packages/typer/__init__.py", line 12, in <module>
    from click.termui import get_terminal_size as get_terminal_size
ImportError: cannot import name 'get_terminal_size' from 'click.termui' (/usr/local/lib/python3.9/site-packages/click/termui.py)
```

JnyJny, 2022-03-28:
Thanks for the bug report! Looks like I got too cavalier about versions of dependencies. Thank you for the workaround and I'll see about getting that fixed.

JnyJny, 2022-03-28:
Looks like this is a [known problem](https://github.com/tiangolo/typer/issues/377) and not necessarily a bug on my part. There is a pull request pending to for typer linked in that issue which fixes the problem.  As soon as it's available I'll push out a release referencing the updated version.

JnyJny, 2022-03-28:
@jsavin could you try installing 0.15.2 of busylight-for-humans and see if that fixes the problem for now?

gavinhungry, 2022-03-29:
@JnyJny I had the same issue as @jsavin with `busylight-for-humans` v0.15.2 from pip (Arch Linux, Python 3.10.4). I also used the same workaround (installing `click` v8.0) successfully.

I removed `click` v8.0 and upgraded `busylight-for-humans` to v0.15.3 from pip, which is working great for now.

Thank you for resolving this so promptly.

JnyJny, 2022-03-29:
Thanks for the report @gavinhungry, I'm always leery to close issues until it's independently confirmed that the resolution is working for people other than me.

closed 2022-03-29 by JnyJny

jsavin, 2022-03-29:
Sorry for the delay: I also tested here with `click` v8.1 and `busylight-for-humans` v 0.15.3 and it worked for me. Thanks for the quick fix!

JnyJny, 2022-03-29:
Thanks for the double confirmation! If only all my bugs were this easy to address :)

tiangolo, 2022-03-30:
I just released Typer `0.4.1` that should handle this. :rocket: :nerd_face:

JnyJny, 2022-03-30:
@tiangolo thank you so much for all you do, big fan of your work.

JnyJny, 2022-03-30:
Published version 0.15.4 with latest dependencies.