JnyJny/busylight #129: Support for Embrava Dim Mode

mattvlasach, 2022-04-05:
Is it possible to support the dim bit for the Embrava lights?  It looks like there is hardware support per the API, but it doesn't look like setting this bit is currently possible.

I'm looking to not glow as much in darker conditions when the light is on :)

mattvlasach, 2022-04-05:
Ha, well I'll just close this one.  I discovered I could just set the second bit of the hex for the relevant color to 0 and it is nicely dimmed by 50%:

red = 0xff0000 (bright) | 0x0f0000 (dim)
green = 0x00ff00 (bright) | 0x000f00 (dim)

JnyJny, 2022-04-28:
Should anyone have an opinion, my proposed implementation is:

```console
$ busylight --dim N [on|blink|fli] [colors]
```

Where `N` is an integer from 1 to 100, representing a scaling value to apply to each channel of the requested color.

JnyJny, 2022-06-20:
The global `--dim` option is now available in version 0.19 of busylight and honored by all the subcommands: on, blink, throb, fli, and rainbow. Additional work done polishing the CLI (catching exceptions, harmonizing global option handling, worked on debug output and other small UX fixes).

```console
$ busylight --dim 100 on green   # no effect, full value green
$ busylight --dim 80 on green    # green is 80% intensity
$ busylight --dim 50 on green    # green is 50% intensity
$ busylight --dim 10 on green    # green is 10% intensity
```

I know this issue is a little stale, but please re-open it if you have any other thoughts about dimming :)

closed 2022-06-20 by JnyJny