JnyJny/busylight #135: Can't stat web API on Ubuntu 22.04

Robindeko, 2022-05-11:
Thanks for this great project!
I can finally use the KUANDO Busylight on my Ubuntu install.

Unfortunately, both the Busylight library and the web API installed correctly (according to PIP), I can't start the Web API.
it keeps on saying "command not found", while the "busylight" command runs without issues.

Does anyone have any tips for me?
Thanks in advance.

Robindeko, 2022-05-11:
Hi JnyJny,

Thank you very much for your quick reply.
It simply says, busyserve: command not found.

[image]

JnyJny, 2022-05-11:
So maybe we should back up. How did you install `busylight-for-humans` and what version is it?

```console
$ python3 -m pip install 'busylight-for-humans[webapi]'
$ python3 -m pip list | grep busy
busylight-for-humans 0.17.2
```
Depending on your shell, the square brackets can cause problems in how the command is interpreted by the shell. Putting them inside the single quotes should keep the shell interpreting them differently then we intended.  For versions 0.16 and earlier, the webapi was started using a subcommand of `busylight`, namely `busylight serve`. For versions 0.17 and later, it's a second command `busyserve`.

JnyJny, 2022-05-11:
Hey! It's a bug in my `pyproject.toml` file, probably from when I merged the master and asyncio branches.

Should have a fix for it pretty quick.

Robindeko, 2022-05-11:
Thanks, good to hear it's probably a bug :)
By the way, when I grep the installed modules from pip, only the busylight-for-humans shows up.

JnyJny, 2022-05-11:
Fixed the pyproject.toml with 09822c2

JnyJny, 2022-05-11:
Ok, version 0.17.3 on pypi should install `busylight` and `busyserve` as expected when installed with the `[webapi]` extras.

Robindeko, 2022-05-11:
awesome, thanks!
I had to run `python3 -m pip install -U busylight-for-humans` to update it to 0.17.3
Works like a charm now.
Thanks again!

closed 2022-05-11 by Robindeko