JnyJny/busylight #158: [BUG] API documentation at `/redoc` is broken

mchaker, 2022-07-16:
**Software Versions:**
 - Operating System: Linux (Ubuntu 22.04)
 - Python version: 3.10.4
 - BusyLight version: 0.21.0

**General Type of Problem**
- [ ] Installation
- [ ] Command-line
- [ ] Integrating into another project
- [x] Web API
- [ ] other

**Describe the Problem**
The `/redoc` route shows a "Loading..." screen and fails to load anything further (the debug output of `busyserve` shows a crash/stack trace)

**Expected Behavior**
Loading and showing the API documentation.

**Error Output**
```
INFO:     100.88.206.101:58997 - "GET /redoc HTTP/1.1" 200 OK
2022-07-16 00:24:24.900 | INFO     | busylight.lights.usblight:all_lights:203 - BlinkStick found 0 lights
2022-07-16 00:24:24.910 | ERROR    | busylight.lights.usblight:acquire:629 - open_path failed with open failed
2022-07-16 00:24:24.911 | ERROR    | busylight.lights.usblight:all_lights:202 - Blynclight LightUnavailable: 2c0d:000c path=b'1-4:1.0'
2022-07-16 00:24:24.912 | INFO     | busylight.lights.usblight:all_lights:203 - Blynclight found 0 lights
2022-07-16 00:24:24.920 | INFO     | busylight.lights.usblight:all_lights:203 - Busylight found 0 lights
2022-07-16 00:24:24.928 | INFO     | busylight.lights.usblight:all_lights:203 - Flag found 0 lights
2022-07-16 00:24:24.935 | INFO     | busylight.lights.usblight:all_lights:203 - Mute found 0 lights
2022-07-16 00:24:24.943 | INFO     | busylight.lights.usblight:all_lights:203 - Orb found 0 lights
2022-07-16 00:24:24.953 | INFO     | busylight.lights.usblight:all_lights:203 - MuteMe found 0 lights
2022-07-16 00:24:24.964 | INFO     | busylight.lights.usblight:all_lights:203 - Status_Indicator found 0 lights
2022-07-16 00:24:24.971 | INFO     | busylight.lights.usblight:all_lights:203 - Blink1 found 0 lights
2022-07-16 00:24:24.972 | INFO     | busylight.lights.usblight:all_lights:195 - USBLight found 0 lights total
INFO:     100.88.206.101:58997 - "GET /openapi.json HTTP/1.1" 500 Internal Server Error
ERROR:    Exception in ASGI application
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/starlette/middleware/base.py", line 43, in call_next
    message = await recv_stream.receive()
  File "/usr/local/lib/python3.10/dist-packages/anyio/streams/memory.py", line 94, in receive
    return self.receive_nowait()
  File "/usr/local/lib/python3.10/dist-packages/anyio/streams/memory.py", line 87, in receive_nowait
    raise EndOfStream
anyio.EndOfStream

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/uvicorn/protocols/http/h11_impl.py", line 403, in run_asgi
    result = await app(self.scope, self.receive, self.send)
  File "/usr/local/lib/python3.10/dist-packages/uvicorn/middleware/proxy_headers.py", line 78, in __call__
    return await self.app(scope, receive, send)
  File "/usr/local/lib/python3.10/dist-packages/fastapi/applications.py", line 269, in __call__
    await super().__call__(scope, receive, send)
  File "/usr/local/lib/python3.10/dist-packages/starlette/applications.py", line 124, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/usr/local/lib/python3.10/dist-packages/starlette/middleware/errors.py", line 184, in __call__
    raise exc
  File "/usr/local/lib/python3.10/dist-packages/starlette/middleware/errors.py", line 162, in __call__
    await self.app(scope, receive, _send)
  File "/usr/local/lib/python3.10/dist-packages/starlette/middleware/base.py", line 68, in __call__
    response = await self.dispatch_func(request, call_next)
  File "/usr/local/lib/python3.10/dist-packages/busylight/api/busylight_api.py", line 205, in light_manager_update
    return await call_next(request)
  File "/usr/local/lib/python3.10/dist-packages/starlette/middleware/base.py", line 46, in call_next
    raise app_exc
  File "/usr/local/lib/python3.10/dist-packages/starlette/middleware/base.py", line 36, in coro
    await self.app(scope, request.receive, send_stream.send)
  File "/usr/local/lib/python3.10/dist-packages/starlette/exceptions.py", line 93, in __call__
    raise exc
  File "/usr/local/lib/python3.10/dist-packages/starlette/exceptions.py", line 82, in __call__
    await self.app(scope, receive, sender)
  File "/usr/local/lib/python3.10/dist-packages/fastapi/middleware/asyncexitstack.py", line 21, in __call__
    raise e
  File "/usr/local/lib/python3.10/dist-packages/fastapi/middleware/asyncexitstack.py", line 18, in __call__
    await self.app(scope, receive, send)
  File "/usr/local/lib/python3.10/dist-packages/starlette/routing.py", line 670, in __call__
    await route.handle(scope, receive, send)
  File "/usr/local/lib/python3.10/dist-packages/starlette/routing.py", line 266, in handle
    await self.app(scope, receive, send)
  File "/usr/local/lib/python3.10/dist-packages/starlette/routing.py", line 65, in app
    response = await func(request)
  File "/usr/local/lib/python3.10/dist-packages/fastapi/applications.py", line 224, in openapi
    return JSONResponse(self.openapi())
  File "/usr/local/lib/python3.10/dist-packages/fastapi/applications.py", line 199, in openapi
    self.openapi_schema = get_openapi(
  File "/usr/local/lib/python3.10/dist-packages/fastapi/openapi/utils.py", line 439, in get_openapi
    return jsonable_encoder(OpenAPI(**output), by_alias=True, exclude_none=True)  # type: ignore
  File "pydantic/main.py", line 341, in pydantic.main.BaseModel.__init__
pydantic.error_wrappers.ValidationError: 1 validation error for OpenAPI
info -> version
  str type expected (type=type_error.str)
```

JnyJny, 2022-07-16:
Well, that is disappointing! I've recreated the error and looking into it. Thanks for reporting this problem.

JnyJny, 2022-07-16:
Good news, found the problem and should have a corrected version published to pypi shortly. Look for version 0.20.6.

mchaker, 2022-07-16:
Excellent, that was quick! Thank you so much.

JnyJny, 2022-07-16:
That should fix the problem, update here if it doesn't.

mchaker, 2022-07-16:
Confirmed fixed - thank you so much!

closed 2022-07-16 by mchaker