JnyJny/busylight #161: [QUESTION] How would I add CORS header `Access-Control-Allow-Origin`?

mchaker, 2022-08-06:
In developing a web frontend for `busylight`, I would like to allow the development machine (running a self-hosted dev version of the web UI) to make requests to the server where the `busylight` API is hosted.

Currently, browsers refuse to make those requests, claiming an `Access-Control-Allow-Origin` header is missing.

Where would I add that header in `busylight_api.py`? I don't think adding the header to every single `GET` endpoint is the right way -- is there some general/middleware layer in which I can add headers to _all_ responses?

I'm willing to make the change and submit a PR -- just not too familiar with where in the API code to add things 😅

mchaker, 2022-08-07:
Update: I think I found the fix.

Now I need to find out whether to take in "accepted origins" from an environment variable (like how HTTP auth credentials are set) or try another approach (perhaps a "debug mode" that allows origins like localhost etc?)

What do you think -- debug mode modification vs. setting origins in an env var? Both?

JnyJny, 2022-08-07:
Well, to be brutally honest, the web API is a proof of concept that I would not feel comfortable deploying in a production environment. I'm not happy with the current implementation since I essentially re-wrote the `busylight.manager.LightManager` class with FastAPI stuff bolted on. Also, I'm not a frontend guy so I have a distinct lack of experience when it comes to implementing web facing stuff. 

After that confession, I am completely open to suggestions as to how to specify the cross origin control to achieve what you are looking for. I like the idea of debug mode being ultra-permissive to avoid configuration issues slowing down development and I am OK with using environment variables to control behavior instead of **_yet another_** configuration file.

I would love for you to submit a PR, I'll review it, have a little back in forth about style probably and then approve it.

mchaker, 2022-08-07:
No worries -- the FastAPI "bolt-on" stuff is pretty good (not that I would have chosen anything different -- I'm not a frontend guy either 😅 I'm just trying to hack together a way to control my lights from my phone)

The solution I found is to add an import for `CORSMiddleware` from FastAPI: https://fastapi.tiangolo.com/tutorial/cors/

I should have a PR out sometime today. 😄

mchaker, 2022-08-07:
@JnyJny Is the `debug` flag available in `busylight_api.py`?

I ask because I am trying to implement the CORS origin allowances for debug mode, but can't find a way to determine if debug mode is set in `busylight_api.py` 🤔

JnyJny, 2022-08-07:
The command line `-D` debug flag is used to enable/disable logging and it's not immediately apparent to me how to pass an argument via the `uvicorn.run` interface.

JnyJny, 2022-08-07:
I think the easiest way to implement this is adding something to the environment before calling uvicorn.run and then checking for it in the api object.

mchaker, 2022-08-07:
> I think the easiest way to implement this is adding something to the environment before calling uvicorn.run and then checking for it in the api object. 

As in adding an `os.environ` value to the runtime environment? (i.e. outside of the `busylight` "program"/executable/script itself -- an actual environment variable?)

Or do you mean storing something in the calling `busylight` class/object? (Not sure how to access the parent/calling object in Python but it seems like it should be possible... `super`? Not sure)

mchaker, 2022-08-07:
Update: I think I figured something out. Now to test my changes... :)

mchaker, 2022-08-07:
@JnyJny after wrestling with dependencies, I can't seem to get my code running. How do I run a development (non-pip-installed) version of the code (e.g. from a copy of the repo/folder on disk?)

JnyJny, 2022-08-08:
I manage dependencies and packaging with the tool [poetry](https://python-poetry.org). You can `pip` install it for convenience:
```console
$ python3 -m pip install poetry
$ cd path/to/busylight
$ poetry shell
<creates a virtual environment with all the dependencies installed>
$ busyserve -D
```

JnyJny, 2022-08-08:
> As in adding an `os.environ` value to the runtime environment? (i.e. outside of the `busylight` "program"/executable/script itself -- an actual environment variable?)
>
> Or do you mean storing [...]

Something like this in `__main__:serve_http_api`:
```
    os.environ["BUSYLIGHT_DISABLE_CORS"] = "True"
    try:
        uvicorn.run("busylight.api:busylightapi", ...)
    except:
        ...
```

And in `api.busylight_api.BusylightAPI`  in the `__init__` check the environment for the existence and value of `BUSYLIGHT_DISABLE_CORS` and do the appropriate action.

It's gross since we are using the environment for "out-of-band" communications when we should just be passing a bool into the the init. When easy things are hard, it's an indication that tools are not being used as they were intended so it's points to some sort of architectural error in the web API code. That's on me of course :)  After the serial integration I'll have to revisit that code to make it less bad.

mchaker, 2022-08-08:
Thanks for the help! :) Yesterday I found a way to grab the `GlobalOptions` object from the "parent"(?) class -- if that doesn't work in my testing, I'll switch to _setting_ separate env vars as you describe in https://github.com/JnyJny/busylight/issues/161#issuecomment-1208215547

mchaker, 2022-08-13:
I attempted to use `poetry` to install dependencies as you described in https://github.com/JnyJny/busylight/issues/161#issuecomment-1208202635 , however I can't seem to run `busyserve` or `busylight` in the `poetry shell` environment. Is there something else I need to configure?

JnyJny, 2022-08-13:
I think I left off an important step in the previous directions:
```
$ python3 -m pip install -U poetry
$ cd path/to/busylight
$ poetry shell
$ poetry install
$ which busylight
```

mchaker, 2022-08-13:
Thanks! That got me further. Hitting a new error:

```
# busyserve -D -p 8888
2022-08-13 16:11:32.859 | INFO     | busylight.__main__:serve_http_api:378 - serving http api
2022-08-13 16:11:32.860 | ERROR    | busylight.__main__:serve_http_api:383 - import uvicorn failed: No module named 'uvicorn'
The package `uvicorn` is missing, unable to serve the busylight API.
```

I'm going to try `python3 -m pip install -U uvicorn`

JnyJny, 2022-08-13:
> Thanks! That got me further. Hitting a new error:
> 
> ```
> # busyserve -D -p 8888
> 2022-08-13 16:11:32.859 | INFO     | busylight.__main__:serve_http_api:378 - serving http api
> 2022-08-13 16:11:32.860 | ERROR    | busylight.__main__:serve_http_api:383 - import uvicorn failed: No module named 'uvicorn'
> The package `uvicorn` is missing, unable to serve the busylight API.
> ```
> 
> I'm going to try `python3 -m pip install -U uvicorn`
> 
> Does the `pyproject.toml` file need to be modified?

Nope, I have helpfully omitted a step. :) The web API part is an optional install target and you can get poetry to install it thusly:

```console
$ poetry shell
$ poetry install --extras webapi  # installs fastapi and uvicorn
$ busyserve -D ...
```

JnyJny, 2022-08-13:
I've updated the top-level README.md to include development installation instructions using poetry. That way its harder for me to forget how it works and mislead people who ask.

mchaker, 2022-08-13:
Haha, thanks! :) I'm about to open a PR for the CORS changes.

PR #163 (2022-08-13): busyserve: add CORS middleware
Closes #161 

# Change description

This change adds support for the CORS `Access-Control-Allow-Origin` header.

That header is required if the API is accessed in a browser, via a host that is not the API host itself. 

For example: if a web UI were developed and hosted on a separate host than where the `busyserve` web API is running.

These changes have been manually tested.

# How to test

Run `busyserve` on host A.
On host B, run a web UI (simple page with javascript making a call to the web API on host A). 
Keep an eye on the browser javascript console: it sh [... cut]
closed 2022-08-13 by JnyJny