JnyJny/busylight #162: [NEW LIGHT] fit-statUSB

mchaker, 2022-08-06:
If you would like BusyLight to support a new device, please provide as much of the following as possible. 

- Product Name: fit-statUSB
- Vendor URL: https://fit-iot.com/web/product/fit-statusb/
- Purchase URL: https://www.amazon.com/gp/product/B07CKFLQ5V
- Public documentation URLs: http://www.fit-pc.com/wiki/index.php/Fit-statUSB

JnyJny, 2022-08-07:
Cool! A new light and thank you for all the useful links. I've ordered an example and should have it shortly. Reviewing the docs, it appears to be a serial device rather than a HID device. I've been working on a re-write to incorporate serial devices side by side with HID lights, but it's still a work in progress with no real ETA. I'm having to re-write the class hierarchy AND learn how serial devices work with Python.

mchaker, 2022-08-07:
Thanks for responding so quickly! If there's any way I can help with the serial device support or testing, let me know. :)

JnyJny, 2022-08-07:
The serial communication part is pretty straight forward, it takes me back to the bad old day of talking to devices on parallel and non-USB serial ports. The light hierarchy is changing from
```
USBLight -> vendor_specific_light
```

to 

```
             +--> HIDLight -> vendor_specific_lights
Light --+
              +--> SerialLight -> vendor_specific_lights
```

So I'm having to find the common ground between HID and serial and move that into `Light` and them implement the protocol specific stuff.  My goal is to have a similar interface that `USBLight` provides so that users don't have to know the implementation specific details of the light they just plugged in:

```python
from busylight.lights import Light

light = Light.first_light()

light.on((255,0,0))
```

JnyJny, 2022-08-07:
I'm working in branch `feature/Light` if you want to follow along, haven't committed anything yet but should have some basic stuff committed in the next 12 hours.

mchaker, 2022-08-07:
I like your approach! I'll follow that branch.

JnyJny, 2022-08-13:
I have a **fit-statUSB** light (and a MuteSync light/button which is also a serial device) and figured out how to drive it using the PySerial package. I'm replacing `busylight.lights.USBLight` and it's hierarchy with `busylight.lights.Light` which has subclasses `HIDLight` and `SerialLight`.

mchaker, 2022-08-13:
Awesome! Thank you so much.

JnyJny, 2022-08-26:
The fit-statUSB is working with the latest commit to the feature/Light branch. It's kind of a limited; it performs fade ins and fade outs for color transitions that are "builtin" and uncommanded. This isn't a big deal for turning the light on and off, but changing between colors quickly looks a little weird. ¯\_(ツ)_/¯   But it's working :) Same goes for the MuteSync serial device.

henryruhs, 2022-09-08:
Check this out for inspiration: https://pypi.org/project/pystatusb

JnyJny, 2022-09-12:
Commit 44053a846a62f0cfff9ee384c7319cedcb58dfb7 adds support for the fit-statUSB and two other lights in addition to a complete restructuring of the Light subclass hierarchy.

mchaker, 2022-09-12:
Excellent!! I'll give it a try this week(end?) 

Thank you so much

henryruhs, 2022-09-12:
I ordered my version of this devices too and will give feedback once it arrived. Thanks Erik

closed 2022-10-02 by JnyJny