JnyJny/busylight #195: [NEW LIGHT]

plaqdk, 2022-10-14:
If you would like BusyLight to support a new device, please provide as much of the following as possible. 

- Product Name: kuando - busylight uc alpha 
- Vendor : plenom

new id just add this to list and it works and thx for the work :) 
(0x27BB, 0x3BCE): "Busylight Alpha",

Best regards o/

JnyJny, 2022-10-14:
Thanks for the updated product information, I've added it to the `busylight.lights.kuando.busylight_alpha.Busylight_Alpha` class which should now recognize your light. I'll push a release shortly and update here when PyPI is updated.

JnyJny, 2022-10-14:
Version 0.24.1 on PyPI now recognizes an additional product id for 'Busylight Alpha`.  I don't anticipate any problems with this change, please re-open this issue if you experience problems recognizing your device.

closed 2022-10-14 by JnyJny