JnyJny/busylight #226: [FEATURE] Adjustment of busylight.color.colortuple_to_name Error Catching

jacobefaust, 2023-02-11:
**Is your feature request related to a problem? Please describe.**
When using _busyserve_ "rainbow" or "pulse" modes where the color is not a standard string ('red', 'blue', 'green', etc.), checking the status via an http://\<ip address\>:8000/lights/status or http://\<ip address\>:8000/light/\<number here\>/status reports an error (per error catching [here](https://github.com/JnyJny/busylight/blob/master/busylight/color.py#L75)).  This error results the json schema to only have "message", but not the other fields typically reported.  This schema variability prevents creation of a custom Home Assistant [RESTful Sensor](https://www.home-assistant.io/integrations/sensor.rest/).

**Describe the solution you'd like**
A modification to error message/catching in `busylight.color.colortuple_to_name` to not report an error message, but instead report some other constant (i.e. "unknown", "null", "error", etc.) in the color field for status calls.

**Describe alternatives you've considered**
Unfortunately, I am not well experienced with http requests (or webdev standards), so I am limited in my recommendations.  I am up for whatever you think is best!

**Additional context**
Here is the status my busylight showed when status was requested in "rainbow" mode:
[image]

JnyJny, 2023-02-11:
Excellent find! That is definitely an oversight on my part.

Instead of raising `ColorLookUpError`, I could just return the hex tuple in string format:

```
def colortuple_to_name(color: Tuple[int, int, int]) -> str:
    """Returns a string name of the given Tuple[int, int, int] if found.

    :color: Tuple[int, int, int]
    :return: str

    Raises:
    - ColorLookupError
    """
    try:
        return webcolors.rgb_to_name(color)
    except ValueError as error:
        logger.info(f"{color} -> {error}")

    r, g, b = color
    return f"#{r:02x}{g:02x}{b:02x}"
```

This should allow the `status` inquiries succeed.

JnyJny, 2023-02-11:
I've tested this commit and I'll publish a patch release with the fix shortly.

jacobefaust, 2023-02-12:
I just pulled in 0.25.2.  It works fantastic!!  Thank you so so much and for the lighting quick fix!

JnyJny, 2023-02-17:
Thanks for confirming that the change works for you!

closed 2023-02-17 by JnyJny