JnyJny/busylight #227: Busylight setup failed to issue with operators

hauenkesyttaja, 2023-02-17:
**Software Versions:**
 - Operating System: Raspberry OS latest
 - Python version: Python Python 3.7.3
 - BusyLight version: latest, just downloaded

**General Type of Problem**
- [x] Installation
- [ ] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [ ] other

**Describe the Problem**
When following the install instructions, on inputing command "busylight udev-rules -o 99-busylights.rules" an error is given

**Expected Behavior**
According to the code and the instruction a file should be outputted for the rules

**Error Output**
$ busylight udev-rules -o 99-busylights.rules
Traceback (most recent call last):

  File "/home/pi/.local/bin/busylight", line 8, in <module>
    sys.exit(cli())

  File "/home/pi/.local/lib/python3.7/site-packages/busylight/__main__.py", line 363, in generate_udev_rules
    print("\n".join(about + rules), file=output)

TypeError: sequence item 2: expected str instance, tuple found

JnyJny, 2023-02-17:
Whoops! Looks like I accidentally committed some feature code into the master branch with the last fix (that obviously doesn't work). Thank you for bringing it to my attention!

JnyJny, 2023-02-17:
I just pushed a new patch revision to PyPI (0.25.3). I'll close this issue after you confirm that the problem is fixed for you.

hauenkesyttaja, 2023-02-17:
Thanks, now the install succeeds without errors. Thank you.

One question though, although it might not be related to this particular issue. When I run the busylight on the command line, it works but does not exit after the task is completed. E.g. if I turn the busylight on with "busylight on" - command the light keeps on but the program does not exit. Then eventually if you need to end the program (crtl + c) then also the light turns off.

So my question or maybe feature request for this command line operation would be to have program which excutes the command e.g. light on but then would exit without turning of the light, to enable next command via the command line for e.g. to change light color.

JnyJny, 2023-02-17:
You are using a Kuando Busylight of some flavor (Alpha, Omega)? Those lights will turn off after a set amount of time (less than 30 seconds IIRC) unless they receive a keep alive message. This means the `busylight` command never exits since it's continually sending the keep alive to the light.  Other supported lights do not have this behavior, it's pretty much unique to the Kuando family.

JnyJny, 2023-02-17:
The `busyserve` command starts an API server that lets you send commands to lights via `curl` or a custom built tool, for example using python and requests. The idea is the API server is already a long lived process that can run in the background and manage sending the keep alive messages while the curl or other method of accessing the API can be short-lived.

JnyJny, 2023-02-17:
I'm going to close this issue since you've confirmed that the original problem is fixed. Please open another issue if you want to discuss other ways to circumvent the questionable design choices made by the Busylight family of devices :)

closed 2023-02-17 by JnyJny