JnyJny/busylight #229: Allow consumer app to adjust log-level

henryruhs, 2023-02-23:
**Is your feature request related to a problem? Please describe.**
The "general" logging implementation does not allow consumer application to adjust the log-level and therfore matching the log-level is impossible.

**Describe the solution you'd like**

1. Introduce a scope for your logger

```python
def get_package_logger() -> Logger:
	return getLogger('busylight')
```

2. Allow to set log-level via public API

```python
def init(log_level : LogLevel) -> None:
	basicConfig(format = '')
	get_package_logger().setLevel(...)
```

**Additional context**
Chroma Feedback has it's own log-level argument that is no longer respected. In this case the INFO should not be printed.

```bash
henry@local:~/chroma-feedback$ ./chroma-feedback -p=microsoft.github -c=thingm.blink1 --log-level=error
2023-02-23 22:07:53.150 | INFO     | busylight.lights.hidlight:available_lights:50 - Blink1 found 1
2023-02-23 22:07:53.150 | ERROR    | busylight.lights.hidlight:acquire:108 - open_path failed: open failed
2023-02-23 22:07:53.150 | ERROR    | busylight.lights.light:first_light:196 - Blink1 b'1-5.4.3.2:1.0'
Connection to thingm.blink1 not found!
Consumer thingm.blink1 crashed!
```

JnyJny, 2023-02-24:
Busylight uses `loguru` for logging which handles things a little differently than logging available in the standard library. I've disabled logging by default in all the busylight packages (which is a best practice I was not aware of).

JnyJny, 2023-02-24:
Patch release 0.25.5 on PyPI has this fix.

closed 2023-02-24 by JnyJny

henryruhs, 2023-02-24:
I think loguru is using logging under the hood. How can people re-enable logging?

JnyJny, 2023-02-24:
I don't know if loguru uses logging under the hood, I suspect the answer is mostly no. You can re-enable logging using:
```python
from loguru import logger

logger.enable('busylight'). # for the entire busylight package
logger.enable('busylight.lights') # just the lights
logger.enable('busylight.effects') # just the effects
logger.enable('busylight.api') # just the web API
````