JnyJny/busylight #232: [BUG] Fit_StatUSB looks broken

henryruhs, 2023-02-28:
Tested on Ubuntu 22 LTS with busylight 0.26.0

```python
#!/usr/bin/env python

from busylight.lights.compulab import Fit_StatUSB

lights = Fit_StatUSB.available_lights()

for light in lights:
	print(light)
	light.on((255, 255, 255))

```

Output:

```bash
(venv) henry@local:~/$ ./test.py
{'vendor_id': 8263, 'product_id': 991, 'device_id': (8263, 991), 'serial_number': '193F914725000600', 'product_string': 'fit_StatUSB - Compulab fit_StatUSB', 'manufacturer_string': 'Compulab LTD', 'path': '/dev/ttyACM0'}
Traceback (most recent call last):
  File "/./test.py", line 10, in <module>
    light.on((255, 255, 255))
AttributeError: 'dict' object has no attribute 'on'
```

JnyJny, 2023-03-01:
The `available_lights` class method is working as documented:

```
 |  available_lights() -> List[Dict[str, Union[bytes, int, str, Tuple[int, int]]]] from abc.ABCMeta
 |      Returns a list of dictionaries describing currently available lights.
 |      
 |      The lights described by those dictionaries may be in use by another process
 |      so may not be "technically" available for use by this process. That question
 |      will be resolved when the Light instance is instantiated
 |      
 |      If the returned list is empty, no supported lights were found.
```

I think the class method you are looking for is `all_lights`:
```
 |  all_lights(reset: bool = True, exclusive: bool = True) -> List[ForwardRef('Light')] from abc.ABCMeta
 |      Returns a list of all lights found in a stable ordering.
 |      
 |      If reset is True, all returned lights are turned off and set
 |      to an inactive state.
 |      
 |      If exclusive is True, all the lights in the list have been
 |      acquired for exclusive use by this process.
 |      
 |      :reset:      bool Quiesce the light when acquired.
 |      :exclusive:  bool Light is owned exclusively by process.
 |      :return: List[Light]
```

henryruhs, 2023-03-01:
Using `all_lights` just returns an empty `[]`. If you can confirm that the driver is working for you, then I assume there is an issue with my Linux, feel free to close this issue.

JnyJny, 2023-03-01:
The `available_lights` class method doesn't attempt to open the device, it's an enumeration of lights that are "available" for use.  If `all_lights` is returning an empty list, it was able to successfully open any of the devices, which points to a permission problem that can be solved with `udev`.  Currently the `busylight udev-rules` command doesn't include rules for serial USB devices, so you would need to write those by hand for now.

henryruhs, 2023-03-01:
So this seems to be related to https://github.com/JnyJny/busylight/issues/173 ... I close that one for now. Thanks

closed 2023-03-01 by henryruhs