JnyJny/busylight #250: [BUG] segmentation fault (core dumped)

rootPound, 2023-04-10:
**Software Versions:**
 - Operating System:  Fedora 37  kernel 6.2.9-200.fc37.x86_64
 - Python version: 3.11.2
 - BusyLight version:  0.26.0

**General Type of Problem**
can't run as normal user

**Describe the Problem**
upgraded my Fedora 37 laptop as normal.  Went to use my busylight, didnt work.  Got this segmentation fault.  Removed the software, reinstalled, followed all the steps for the udev permissions.  Still can't access as a normal user.  I can run busylight as root with sudo.   

**Error Output**

Dmesg output....

[92906.887607] usb 3-1.3: USB disconnect, device number 4
[92915.059509] usb 3-1.3: new full-speed USB device number 11 using xhci_hcd
[92915.212480] usb 3-1.3: New USB device found, idVendor=2c0d, idProduct=000c, bcdDevice= 0.01
[92915.212500] usb 3-1.3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[92915.212509] usb 3-1.3: Product: Blynclight Standard
[92915.222004] hid-generic 0003:2C0D:000C.000C: ignoring exceeding usage max
[92915.222602] hid-generic 0003:2C0D:000C.000C: ignoring exceeding usage max
[92915.224109] hid-generic 0003:2C0D:000C.000C: hiddev98,hidraw2: USB HID v1.10 Device [Blynclight Standard] on usb-0000:3c:00.0-1.3/input0

Error in dmesg when I run as normal user...

[92925.239050] traps: busylight[23069] general protection fault ip:7f440ce0b345 sp:7ffc31acbcf0 error:0 in libusb-1-150b88da.0.so.0.1.0[7f440ce00000+17000]

Udev rule in the /etc/udev/rules.d directory....

# 2 Embrava Blynclight
KERNEL=="hidraw*", ATTRS{idVendor}=="2c0d", ATTRS{idProduct}=="000c", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="2c0d", ATTRS{idProduct}=="000c", MODE="0666"

JnyJny, 2023-04-10:
Thanks for the bug report. I suspect the problem might be the updated `udev` subsystem for Linux might not be as forgiving of formatting of the vendor and product identifiers. Could you edit the udev rules and prepend 0x to the vendor and product identifiers, like this:

```
KERNEL=="hidraw*", ATTRS{idVendor}=="0x2c0d", ATTRS{idProduct}=="0x000c", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0x2c0d", ATTRS{idProduct}=="0x000c", MODE="0666"
```

And let me know if that fixes the problem?

rootPound, 2023-04-10:
Just gave it try.  after updating the udev file I ran "udevadmn control -R".   unplug/plugged back in.   As a normal user I no longer get the segmentation fault error but I do now get "No lights to turn on".   No further error message in dmesg.

JnyJny, 2023-04-10:
Well, that's a step in the right direction. Someone in libusb or the udev subsystem might like to know about text configuration files inducing a crash (instead of something a little more civilized like an error message and crash :).

I'll have to go back and read up on udev to see what's going on.

rootPound, 2023-04-10:
No worries!  at least sudo works for now.   Greatly appreciate it!

closed 2023-08-07 by JnyJny