JnyJny/busylight #283: [BUG]

adambier, 2023-07-22:
**Software Versions:**
 - Operating System:  MacOS Sonoma 14.0 public beta 1
 - Python version: 3.9.6
 - BusyLight version: 0.26.0

**General Type of Problem**
- [ ] Installation
- [ ] Command-line
- [ ] Integrating into another project
- [X ] Web API
- [ ] other

**Describe the Problem**
After setting all lights to flicker with http://localhost:8000/lights/fli, http://localhost:8000/lights/off doesn't work.

**Expected Behavior**
http://localhost:8000/lights/off should turn off the flickering light

**Error Output**
Instead, it does nothing. In order to turn off the flickering light, I have to first http://localhost:8000/lights/on and then http://localhost:8000/lights/off

JnyJny, 2023-07-25:
Thanks for the bug report and I'm sorry it's taken this long to respond!  I haven't been able to reproduce the bug yet but I am seeing a server hang in any of the animated modes (fli, blink & rainbow) which may or may not be related. Could you tell me what specific lights you have connected and how many of each?  That will help me replicate your situation more closely.

adambier, 2023-08-02:
No worries, thanks! I've got both a Kuando Busylight Omega and a Blink(1) (mk3), and the problem exists whether I have either or both attached.

One interesting note is that you may not always see this problem occur in Safari because as you start to edit the URL to turn the lights off, Safari begins to autofill and preload the page once you get to "http://localhost:8000/lights/o", preloading "http://localhost:8000/lights/on" before you replace the "n" with "ff".

JnyJny, 2023-08-02:
I've replicated the problem (without the hang) with just a blink(1) attached and I've got some time this weekend so hopefully I'll get this fixed shortly.

JnyJny, 2023-08-05:
I think I've got a fix, there was a `task_cancel` missing in the code that turns the lights off in the FastAPI class.

JnyJny, 2023-08-06:
Published version 0.26.1 to PyPI which has the fix, give it a try and see if you can get it to break.

adambier, 2023-08-06:
It's working properly for me now (with both the blink(1) and the busylight omega, or either)--thanks!

closed 2023-08-07 by JnyJny