JnyJny/busylight #295: [Help Request] Blink

debistevens, 2023-09-12:
I'm trying to update a busylight script I have from an old version of busylight-for-humans==0.15.0.  I'm a hobbyist and would appreciate some help for things I'm not finding in the documentation. 

It's a simple script that if it detects the microphone in use icon in the tray, then I want it to make the light blink red/off.

With that version, I was able to accomplish that with something along the lines of

```
from busylight.lights.embrava import Blynclight

LIGHT = Blynclight.first_light()

RED = (255, 0, 0)
BLINK = 1

if mic_status:
    LIGHT.blink(RED, BLINK)
```

I dug through the various releases and notes, but I've not been able to parse how to use the new import in conjunction with my light.  first_light() still turns it on, but I cannot figure out how to make it blink with 
```
from busylight.effects import Blink
```

Thank you.

JnyJny, 2023-09-13:
Yeah, that was my fault. I removed `blink` as a generic `busylight.lights.light.Light` method because several of the lights I was trying to support didn't have a blink mode that didn't require software intervention.  And then I re-wrote it busylight-for-humans again to support the effects you found via asyncio instead of the previous threaded version. It's ok if none of this makes sense.  The _easiest_ way to get your light blinking when you want is using the `busyserve` command:

```console
$ busyserve >& busyserve.log &    # start the busylight server in the background and keep the log
```

Next, talk to your light, or lights, using HTTP:

```python
# make the first light found blink for 10 seconds and then turn off

from time import sleep
import requests

response = request.get("http://localhost:8000/lights")

lights = response.json()

light_id = lights[0]['light_id']

response = request.get(f"http://localhost:8000/light/{light_id}/blink")

print(response.json())

sleep(10)

response = request.get(f"http://localhost:8000/light/{light_id}/off")

print(response.json())
```

I know it's more involved than the version you were using, I was trying to normalize how lights behaved and none of them knew how to blink very well (except the Embrava Busylight family).

Let me know how that works for you and I'll consider re-adding the `blink` method to the `Light` class.

debistevens, 2023-09-14:
Hi JnyJny,

Thanks for the detailed explanation!  

I was able to make some progress using the manager per the README.md, although the Python script would get hung up at applying the Blink effect.  I need to configure logging to see if I can identify why it's hanging at that line, whenever I get some time.

I went super simple and just decided to toggle it off/on right now with some sleep commands, but will also look into the blinkserve/api as I think that will expand my use cases.  

For my own understanding, is there some more documentation or code samples with using the manager or combining the Blynclight.first_light() and the Effects classes?

I appreciate your help!

JnyJny, 2023-09-14:
The best example I have for using the Effects classes is in `busylight/__main__.py`.  The `blink` subcommand for the command line interface is implemented as:

```python
@cli.command(name="blink")
def blink_lights(
    ctx: typer.Context,
    color: Optional[str] = typer.Argument(
        "red", callback=string_to_scaled_color, show_default=True
    ),
    speed: Speed = typer.Argument(
        Speed.Slow,
        show_default=True,
    ),
) -> None:
    """Blink light on and off."""
    logger.info("blinking lights")

    blink = Effects.for_name("blink")(color, speed.duty_cycle)

    try:
        manager.apply_effect(blink, ctx.obj.lights, timeout=ctx.obj.timeout)
    except (KeyboardInterrupt, TimeoutError):
        manager.off(ctx.obj.lights)
    except NoLightsFound as error:
        typer.secho("Unable to blink lights.", fg="red")
        raise typer.Exit(code=1) from None
```

The `manager` refers to an instance of `busylight.manager.LightManager` which mediates the discovery of lights and applying effects to them via asyncio.

debistevens, 2023-09-15:
Great, I'll work on this some more this weekend.  No need to re-implement the the blink method to the Light class.  Thank you!

closed 2023-09-15 by debistevens