JnyJny/busylight #301: No attribute

patrick-dojofive, 2023-11-21:
**Software Versions:**
 - Operating System: MacOS 14.1
 - Python version: 3.11.6
 - BusyLight version: 0.26.1

**General Type of Problem**
- [ ] Installation
- [x] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [ ] other

**Describe the Problem**
After installation, running simple commands results in an error

**Error Output**
```sh
Successfully installed busylight-for-humans-0.26.1
❯ busylight on red
Traceback (most recent call last):
  File "/opt/homebrew/bin/busylight", line 5, in <module>
    from busylight.__main__ import cli
  File "/opt/homebrew/lib/python3.11/site-packages/busylight/__main__.py", line 14, in <module>
    from .lights import NoLightsFound
  File "/opt/homebrew/lib/python3.11/site-packages/busylight/lights/__init__.py", line 56, in <module>
    from .hidlight import HIDLight
  File "/opt/homebrew/lib/python3.11/site-packages/busylight/lights/hidlight.py", line 15, in <module>
    class HIDLight(Light):
  File "/opt/homebrew/lib/python3.11/site-packages/busylight/lights/hidlight.py", line 79, in HIDLight
    def device(self) -> hid.device:
                        ^^^^^^^^^^
AttributeError: module 'hid' has no attribute 'device'. Did you mean: 'Device'?
```

JnyJny, 2023-11-21:
First, thanks for the bug report!

I haven't been able to recreate this error yet; MacOS 14.1, python 3.11.6 and busylight installed using pip or from the repo.

```console
$ python3
Python 3.11.6 (main, Nov 21 2023, 16:15:44) [Clang 15.0.0 (clang-1500.0.40.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> hid.device
<class 'hid.device'>
>>> hid.Device
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'hid' has no attribute 'Device'. Did you mean: 'device'?
```

I guess I'm worried that the `hid` module installed isn't `hidapi` version 0.14.0 and is some other hid package?

```
$ python3 -m pip list
Package              Version
-------------------- -------
bitvector-for-humans 0.14.1
busylight-for-humans 0.26.1
click                8.1.7
hidapi               0.14.0
loguru               0.7.2
pip                  23.3.1
pyserial             3.5
setuptools           65.5.0
typer                0.9.0
typing_extensions    4.8.0
webcolors            1.13
```

Can you give me a `pip list` of the packages installed with busylight and describe how you installed busylight?

patrick-dojofive, 2023-11-22:
```python
❯ python3
Python 3.11.6 (main, Oct  2 2023, 13:45:54) [Clang 15.0.0 (clang-1500.0.40.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> hid.device
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'hid' has no attribute 'device'. Did you mean: 'Device'?
>>> hid.Device
<class 'hid.Device'>
```
and
```python
❯ pip list
Package              Version
-------------------- ------------
annotated-types      0.6.0
anyio                4.0.0
bitvector-for-humans 0.14.1
blinker              1.7.0
build                1.0.3
busylight-for-humans 0.26.1
busylite             0.1.1
CacheControl         0.13.1
certifi              2023.7.22
cffi                 1.16.0
charset-normalizer   3.3.2
cleo                 2.1.0
click                8.1.7
crashtest            0.4.1
distlib              0.3.7
docutils             0.20.1
dulwich              0.21.6
fastapi              0.100.1
fastjsonschema       2.19.0
filelock             3.13.1
Flask                3.0.0
Flask-HTTPAuth       4.8.0
h11                  0.14.0
hid                  1.0.5
hidapi               0.14.0
idna                 3.4
importlib-metadata   6.8.0
installer            0.7.0
itsdangerous         2.1.2
jaraco.classes       3.3.0
Jinja2               3.1.2
keyring              24.3.0
loguru               0.7.2
MarkupSafe           2.1.3
more-itertools       10.1.0
msgpack              1.0.7
numpy                1.26.2
packaging            23.2
pexpect              4.8.0
pip                  23.3.1
pkginfo              1.9.6
platformdirs         3.11.0
poetry               1.7.1
poetry-core          1.8.1
poetry-plugin-export 1.6.0
ptyprocess           0.7.0
pycparser            2.21
pydantic             2.5.1
pydantic_core        2.14.3
pyproject_hooks      1.0.0
pyserial             3.5
rapidfuzz            3.5.2
requests             2.31.0
requests-toolbelt    1.0.0
setuptools           68.2.2
shellingham          1.5.4
six                  1.16.0
sniffio              1.3.0
starlette            0.27.0
tomlkit              0.12.3
trove-classifiers    2023.11.14
typer                0.9.0
typing_extensions    4.8.0
urllib3              2.1.0
uvicorn              0.24.0.post1
virtualenv           20.24.7
webcolors            1.13
Werkzeug             3.0.1
wheel                0.41.3
xattr                0.10.1
zipp                 3.17.0
```

I see I have both hid and hidapi modules. I'll dig there.

patrick-dojofive, 2023-11-22:
Yep that's it...

patrick-dojofive, 2023-11-22:
This has come up before elsewhere now that I know what I'm searching for.
https://github.com/qmk/qmk_cli/issues/82

JnyJny, 2023-11-22:
I agree that QMK is seeing the same problem, it looks like they decided to punt and neither of the projects seemed inclined to step up as far as de-conflicting the namespace. Maybe there are some guards I can put in place to warn about package mismatches (or even shim so that it "just works"™).

JnyJny, 2023-11-24:
The updated HID handling is in version 0.27.0 and published to PyPI. If you're so inclined, you could pip install it, install hid and confirm that busylight continues to work.

patrick-dojofive, 2023-11-27:
Installed `hid`, reconfirmed original observation, upgraded `busylight-for-humans`, tested and confirmed the upgrade handles the namespace collision.
```
❯ pip install hid
Collecting hid
  Using cached hid-1.0.5-py3-none-any.whl
Installing collected packages: hid
Successfully installed hid-1.0.5
❯ python3
Python 3.11.6 (main, Oct  2 2023, 13:45:54) [Clang 15.0.0 (clang-1500.0.40.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> hid.device
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'hid' has no attribute 'device'. Did you mean: 'Device'?
>>> hid.Device
<class 'hid.Device'>
>>> ^D
❯ pip install --upgrade busylight-for-humans
Requirement already satisfied: busylight-for-humans in /opt/homebrew/lib/python3.11/site-packages (0.26.1)
Collecting busylight-for-humans
  Downloading busylight_for_humans-0.27.6-py3-none-any.whl.metadata (10 kB)
Requirement already satisfied: bitvector-for-humans<0.15.0,>=0.14.0 in /opt/homebrew/lib/python3.11/site-packages (from busylight-for-humans) (0.14.1)
Requirement already satisfied: hidapi<0.15.0,>=0.14.0 in /opt/homebrew/lib/python3.11/site-packages (from busylight-for-humans) (0.14.0)
Requirement already satisfied: loguru<0.8,>=0.6 in /opt/homebrew/lib/python3.11/site-packages (from busylight-for-humans) (0.7.2)
Requirement already satisfied: pyserial<4.0,>=3.5 in /opt/homebrew/lib/python3.11/site-packages (from busylight-for-humans) (3.5)
Requirement already satisfied: typer<0.10,>=0.7 in /opt/homebrew/lib/python3.11/site-packages (from busylight-for-humans) (0.9.0)
Requirement already satisfied: webcolors<2.0.0,>=1.11.1 in /opt/homebrew/lib/python3.11/site-packages (from busylight-for-humans) (1.13)
Requirement already satisfied: setuptools>=19.0 in /opt/homebrew/lib/python3.11/site-packages (from hidapi<0.15.0,>=0.14.0->busylight-for-humans) (68.2.2)
Requirement already satisfied: click<9.0.0,>=7.1.1 in /opt/homebrew/lib/python3.11/site-packages (from typer<0.10,>=0.7->busylight-for-humans) (8.1.7)
Requirement already satisfied: typing-extensions>=3.7.4.3 in /opt/homebrew/lib/python3.11/site-packages (from typer<0.10,>=0.7->busylight-for-humans) (4.8.0)
Downloading busylight_for_humans-0.27.6-py3-none-any.whl (55 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 55.3/55.3 kB 2.9 MB/s eta 0:00:00
Installing collected packages: busylight-for-humans
  Attempting uninstall: busylight-for-humans
    Found existing installation: busylight-for-humans 0.26.1
    Uninstalling busylight-for-humans-0.26.1:
      Successfully uninstalled busylight-for-humans-0.26.1
Successfully installed busylight-for-humans-0.27.6
❯ busylight --timeout 10 fli

JnyJny, 2023-11-27:
Thank you very much for your help!

closed 2023-11-27 by JnyJny