JnyJny/busylight #306: Blynclight Plus not detected

loic-gouraud, 2023-12-01:
Hello
Sorry for my English, I'm French.
I encounter a detection problem with my Blynclight Plus: No lights detected.
Obviously there is something strange:

**root@raspicube:~# lsusb**
Bus 001 Device 004: ID 2c0d:0002 Blynclight Plus

**root@raspicube:~# busylight -D list**
2023-12-01 14:53:42.262 | INFO     | busylight.__main__:precommand_callback:132 - version 0.27.6
2023-12-01 14:53:42.263 | INFO     | busylight.__main__:precommand_callback:133 - timeout=None
2023-12-01 14:53:42.263 | INFO     | busylight.__main__:precommand_callback:134 -     dim=1.0
2023-12-01 14:53:42.264 | INFO     | busylight.__main__:precommand_callback:135 -  lights=[]
2023-12-01 14:53:42.265 | INFO     | busylight.__main__:precommand_callback:136 -     cmd='list'
2023-12-01 14:53:42.268 | INFO     | busylight.__main__:list_available_lights:275 - listing connected lights
...
2023-12-01 14:53:42.301 | INFO     | busylight.lights.hidlight:available_lights:52 - Blynclight_Plus found 0
2023-12-01 14:53:42.302 | INFO     | busylight.lights.light:all_lights:160 - Blynclight_Plus found 0 lights.
..
2023-12-01 14:53:42.414 | INFO     | busylight.lights.light:all_lights:153 - Light found 0 lights total.
No lights detected.

can you help me ?

JnyJny, 2023-12-05:
Hello and thank you for the bug report! I was at first concerned that you didn't have the appropriate permission for the device via the udev subsystem, however your example output shows you running with root already so that theory was at the door. Thank you for providing the output of `lsusb` , that allowed me to determine that `0x2c0d:0x0020` is an unknown (to me) vendor/product identifier for Blynclight Plus.  I've added support for that product id and published a new version to PyPI, 0.27.7. Please let me know if it fixes the problem for you!

loic-gouraud, 2023-12-06:
Hello, it's ok with this 0.28 now. thank you so much.

closed 2023-12-06 by loic-gouraud