JnyJny/busylight #312: [BUG] ERROR: Could not build wheels for hidapi, which is required to install pyproject.toml-based projects

Swahgner, 2023-12-22:
**Software Versions:**
 - Operating System: Raspbian GNU/Linux 12 (bookworm)
 - Python version: 3.11.2
 - BusyLight: Kuando Omega

**General Type of Problem**
- [ X ] Installation
- [ ] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [ ] other

**Describe the Problem**
When running `python3 -m pip install busylight-for-humans`, I get the error in the title

**Expected Behavior**
I expect it to install...

**Error Output**
`Building wheels for collected packages: hidapi
  Building wheel for hidapi (setup.py) ... error
  error: subprocess-exited-with-error

  python setup.py bdist_wheel did not run successfully.
  exit code: 1

  [19 lines of output]
  /home/relliot/.local/lib/python3.11/site-packages/setuptools/__init__.py:80: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
  !!

          ********************************************************************************
          Requirements should be satisfied by a PEP 517 installer.
          If you are using pip, you can try `pip install --use-pep517`.
          ********************************************************************************

  !!
    dist.fetch_build_eggs(dist.setup_requires)
  running bdist_wheel
  running build
  running build_ext
  Compiling hid.pyx because it changed.
  [1/1] Cythonizing hid.pyx
  /home/relliot/.local/lib/python3.11/site-packages/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /tmp/pip-install-85i8kzpk/hidapi_c909f4a410e14dce968b40b1b7755855/hid.pyx
    tree = Parsing.p_module(s, pxd, full_module_name)
  building 'hid' extension
  error: unknown file type '.pxd' (from 'chid.pxd')
  [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for hidapi
  Running setup.py clean for hidapi
Failed to build hidapi
ERROR: Could not build wheels for hidapi, which is required to install pyproject.toml-based projects`

JnyJny, 2023-12-22:
Thanks for the bug report! Unfortunately, the source of that error is the [hidapi][0] package.  The install process couldn't find a prebuilt `hidapi` package for your operating system and CPU, so it attempted to build it from source and failed. This could be due to missing or down level software used to build the hidapi package (Cython I think). I've been able to install Busylight successfully on a Raspberry Pi in the past but haven't tested it recently.

[0]: https://github.com/trezor/cython-hidapi

JnyJny, 2023-12-22:
This [Stackoverflow question][0] has a solution for a similar situation, maybe it will help you.

[0]: https://stackoverflow.com/questions/34654173/unknown-file-type-error-with-pxd-file

Swahgner, 2024-01-03:
> This [Stackoverflow question](https://stackoverflow.com/questions/34654173/unknown-file-type-error-with-pxd-file) has a solution for a similar situation, maybe it will help you.

Hey, thanks for giving it a shot.

Unfortunately, it didn't work out. I've spent hours trying to fix it, but no luck. As you mentioned, it seems to be a hidapi problem, and I might reach out to them and see if they can help.

Jaharmi, 2024-01-07:
@Swahgner Which kind of Raspberry Pi are you installing on? I've always installed on a Raspberry Pi … but also not recently.

I forget which model I'm using currently, but it's probably something like a Raspberry Pi 3.

I was _not_ able to install on a Raspberry Pi Zero W back in the day. I seem to recall this was some issue involving Rust, which may have been due to a dependency. My memory is fuzzy on this, however. There may have been an issue about it.

Swahgner, 2024-01-08:
> @Swahgner Which kind of Raspberry Pi are you installing on? I've always installed on a Raspberry Pi … but also not recently.
> 
> I forget which model I'm using currently, but it's probably something like a Raspberry Pi 3.
> 
> I was _not_ able to install on a Raspberry Pi Zero W back in the day. I seem to recall this was some issue involving Rust, which may have been due to a dependency. My memory is fuzzy on this, however. There may have been an issue about it.
> 
> **Update:** Yeah, it was #2 and it was an issue with a FastAPI dependency.

I have tried on a Raspberry Pi 3 Model A+. I tried switching to an ubuntu server OS instead, but everything leads back to hidapi failing to build wheels. I have tried countless of "solutions" I've found, but everytime it's the same issue

closed 2024-05-13 by JnyJny