JnyJny/busylight #356: [BUG] USB to Serial Device Misidentified as Busylight when it is not

jake-vu, 2024-05-23:
**Software Versions:**
 - Operating System: Linux 6.7.4-arch1-1 x86_64 GNU/Linux
 - Python version: 3.11.7
 - BusyLight version: 0.30.0

**General Type of Problem**
- [ ] Installation
- [ ] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [x] other

**Describe the Problem**
I have a custom board using a SiliconLabs CP2102 USB to Serial Converter.  The unrelated device is misidentified as a MuteSync busylight.   

Running `lsusb -v -s 1:9` to show the custom, unrelated device details:
```
Bus 001 Device 009: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x10c4 Silicon Labs
  idProduct          0xea60 CP210x UART Bridge
  bcdDevice            1.00
  iManufacturer           1 Vutility
  iProduct                2 VoltDrop Calibration Cradle
  iSerial                 3 VDC01
  bNumConfigurations      1
```

After launching busyserve and running `curl -s http://localhost:8000/lights/status | jq` 
```
[
  {
    "light_id": 0,
    "name": "Busylight Omega",
    "info": {
      "path": "1-6.1:1.0",
      "vendor_id": 10171,
      "product_id": 15309,
      "serial_number": "",
      "release_number": 256,
      "manufacturer_string": "PLENOM APS",
      "product_string": "BUSYLIGHT",
      "usage_page": 0,
      "usage": 0,
      "interface_number": 0,
      "device_id": [
        10171,
        15309
      ]
    },
    "is_on": false,
    "color": "black",
    "rgb": [
      0,
      0,
      0
    ]
  },
  {
    "light_id": 1,
    "name": "MuteSync Button",
    "info": {
      "vendor_id": 4292,
      "product_id": 60000,
      "device_id": [
        4292,
        60000
      ],
      "serial_number": "VDC01",
      "product_string": "VoltDrop Calibration Cradle",
      "manufacturer_string": "Vutility",
      "path": "/dev/ttyUSB0"
    },
    "is_on": false,
    "color": "black",
    "rgb": [
      0,
      0,
      0
    ]
  },
  {
    "light_id": 2,
    "name": "MuteSync Button",
    "info": {
      "vendor_id": 4292,
      "product_id": 60000,
      "device_id": [
        4292,
        60000
      ],
      "serial_number": "VDC01",
      "product_string": "VoltDrop Calibration Cradle",
      "manufacturer_string": "Vutility",
      "path": "/dev/ttyUSB0"
    },
    "is_on": false,
    "color": "black",
    "rgb": [
      0,
      0,
      0
    ]
  }
]
```

**Expected Behavior**
SiLabs 2102x devices which are not Mutesync do not appear as busylights and do not send USB communication. Though vid and pid may match, the product and manufacturer strings should would also be checked to match exactly. 

**Error Output**
N/A

jake-vu, 2024-05-23:
Thank you for your work on this repository! For now, I have simply commented out the entry in the dict return of MuteSync::suported_device_ids() and that seems to have fixed it. 

With what you were saying about devices exporting valid/useful manufacturer or product strings: I did a bit more research and it appears that the MuteSync folks did not change the CP2102 from the default Manufacturer Name, which still appears as Silicon Labs. Per their website FAQ:
```
    Instance ID: USB\VID_10C4&PID_EA60\S7GWBRO3JUH8AF 

    Device Description: Silicon Labs CP210x USB to UART Bridge (COM4) 

    Class Name: Ports 

    Class GUID: {4d36e978-e325-11ce-bfc1-08002be10318} 

    Manufacturer Name: Silicon Labs 

    Status: Started 

    Driver Name: oem37.inf 

    Hardware IDs: 

         USB\VID_10C4&PID_EA60&REV_0100

         USB\VID_10C4&PID_EA60 

    Compatible IDs: 

         USB\Class_FF&SubClass_00&Prot_00 

         USB\Class_FF&SubClass_00 

         USB\Class_FF
```  

So there may not be anything that you can do in code without impacting other devices which also use that SiLabs part and have not updated their Manufacturer/Product strings.

JnyJny, 2024-05-24:
I've extended the logic around how MuteSync devices are claimed so that it should not attempt to control your device. The new version is published to PyPI as version 0.30.1. Let me know if that fixes the problem for you.  After installing, the `busylight` command-line interface should be available and you can use it to list devices that busylight recognizes:

```
$ python3 -m pip install -U busylight-for-humans
...
$ busylight list -v
```