JnyJny/busylight #417: Windows requires byte prepended to data

JnyJny, 2024-11-26:
Also, as a separate note, I had to deploy on Windows and to make it work I had to send an additional 0 to the devices. Then everything was working; without it the lights were ignoring any commands. I have no idea why it works with the 0.
And I had 2 different devices, a Luxafor ORB and a Kuando, and both behaved the same.

```python
if "WINDOWS" in platform.system().upper():

    def patched_write_strategy(instance: lightslib.Light, data: bytes) -> int:
        data = bytes([0]) + data
        return instance.device.write(data)

    lightslib.Light.write_strategy = patched_write_strategy
```

_Originally posted by @aponoran in https://github.com/JnyJny/busylight/issues/416#issuecomment-2501453875_

JnyJny, 2024-11-26:
@aponoran  I created a new issue to track the Windows extra byte problem. 

As I mentioned in #416, we need to preserve the `write_strategy` property which encodes which write method a device needs: write or send_feature_report.  I mentioned making the windows check in the `__bytes__` magic method but I think it makes more sense to check for windows in the method: `busylight.lights.light.Light.update`. I'm going to commit an update in a second with a terrible first pass check for Windows.

JnyJny, 2024-11-29:
Possible fix for #87 and #56.

aponoran, 2024-12-01:
I tested on Windows with the above commit and everything works fine! Used a Luxafor ORB and a Kuando Omega.

JnyJny, 2024-12-03:
This fix is now available in version 0.33.0. Again, thanks for your help!

closed 2024-12-03 by JnyJny