JnyJny/busylight #434: [BUG]

EddiesTech, 2025-02-24:
**Software Versions:**
 - Operating System: Windows 11 23H2 (v10.0.22631)
 - Python version: 3.13.2
 - BusyLight version: 0.33.2 (installed via pip)

**General Type of Problem**
- [ ] Installation
- [x] Command-line
- [ ] Integrating into another project
- [x] Web API
- [ ] other

**Describe the Problem**
Plantronics Status Indicator remains blank even when sending commands. Status Indicator is found, but remains blank. I have Embrava Connect installed, but disabled. Same via CLI or Web API.

**Expected Behavior**
Status Indicator should display the light/colour that was in the command.

**Error Output**
No error, but here is the debug log when running "on" command
```
C:\Windows\System32>busylight -D on
2025-02-24 14:26:30.317 | INFO     | busylight.__main__:precommand_callback:132 - version 0.33.2
2025-02-24 14:26:30.317 | INFO     | busylight.__main__:precommand_callback:133 - timeout=None
2025-02-24 14:26:30.318 | INFO     | busylight.__main__:precommand_callback:134 -     dim=1.0
2025-02-24 14:26:30.318 | INFO     | busylight.__main__:precommand_callback:135 -  lights=[0]
2025-02-24 14:26:30.318 | INFO     | busylight.__main__:precommand_callback:136 -     cmd='on'
2025-02-24 14:26:30.320 | INFO     | busylight.__main__:turn_lights_on:149 - activating lights
2025-02-24 14:26:30.396 | INFO     | busylight.lights.hidlight:available_lights:52 - BlinkStick found 0
2025-02-24 14:26:30.397 | INFO     | busylight.lights.light:all_lights:149 - BlinkStick found 0 lights.
2025-02-24 14:26:30.488 | INFO     | busylight.lights.hidlight:available_lights:52 - Blynclight found 0
2025-02-24 14:26:30.489 | INFO     | busylight.lights.light:all_lights:149 - Blynclight found 0 lights.
2025-02-24 14:26:30.577 | INFO     | busylight.lights.hidlight:available_lights:52 - Blynclight_Mini found 0
2025-02-24 14:26:30.577 | INFO     | busylight.lights.light:all_lights:149 - Blynclight_Mini found 0 lights.
2025-02-24 14:26:30.648 | INFO     | busylight.lights.hidlight:available_lights:52 - Blynclight_Plus found 0
2025-02-24 14:26:30.649 | INFO     | busylight.lights.light:all_lights:149 - Blynclight_Plus found 0 lights.
2025-02-24 14:26:30.729 | INFO     | busylight.lights.hidlight:available_lights:52 - Status_Indicator found 1
2025-02-24 14:26:30.735 | INFO     | busylight.lights.hidlight:acquire:109 - Status Indicator open_path(b'\\\\?\\HID#VID_047F&PID_D005#a&24b118b7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}') succeeded
2025-02-24 14:26:30.735 | INFO     | busylight.lights.light:all_lights:149 - Status_Indicator found 1 lights.
2025-02-24 14:26:30.813 | INFO     | busylight.lights.hidlight:available_lights:52 - Busylight_Alpha found 0
2025-02-24 14:26:30.814 | INFO     | busylight.lights.light:all_lights:149 - Busylight_Alpha found 0 lights.
2025-02-24 14:26:30.902 | INFO     | busylight.lights.hidlight:available_lights:52 - Busylight_Omega found 0
2025-02-24 14:26:30.902 | INFO     | busylight.lights.light:all_lights:149 - Busylight_Omega found 0 lights.
2025-02-24 14:26:30.985 | INFO     | busylight.lights.hidlight:available_lights:52 - Flag found 0
2025-02-24 14:26:30.986 | INFO     | busylight.lights.light:all_lights:149 - Flag found 0 lights.
2025-02-24 14:26:31.070 | INFO     | busylight.lights.hidlight:available_lights:52 - Bluetooth found 0
2025-02-24 14:26:31.070 | INFO     | busylight.lights.light:all_lights:149 - Bluetooth found 0 lights.
2025-02-24 14:26:31.151 | INFO     | busylight.lights.hidlight:available_lights:52 - Mute found 0
2025-02-24 14:26:31.151 | INFO     | busylight.lights.light:all_lights:149 - Mute found 0 lights.
2025-02-24 14:26:31.232 | INFO     | busylight.lights.hidlight:available_lights:52 - Orb found 0
2025-02-24 14:26:31.232 | INFO     | busylight.lights.light:all_lights:149 - Orb found 0 lights.
2025-02-24 14:26:31.319 | INFO     | busylight.lights.hidlight:available_lights:52 - MuteMe found 0
2025-02-24 14:26:31.320 | INFO     | busylight.lights.light:all_lights:149 - MuteMe found 0 lights.
2025-02-24 14:26:31.405 | INFO     | busylight.lights.hidlight:available_lights:52 - MuteMe_Mini found 0
2025-02-24 14:26:31.406 | INFO     | busylight.lights.light:all_lights:149 - MuteMe_Mini found 0 lights.
2025-02-24 14:26:31.493 | INFO     | busylight.lights.hidlight:available_lights:52 - Blink1 found 0
2025-02-24 14:26:31.493 | INFO     | busylight.lights.light:all_lights:149 - Blink1 found 0 lights.
2025-02-24 14:26:31.504 | INFO     | busylight.lights.seriallight:available_lights:83 - Fit_StatUSB found 0
2025-02-24 14:26:31.505 | INFO     | busylight.lights.light:all_lights:149 - Fit_StatUSB found 0 lights.
2025-02-24 14:26:31.512 | INFO     | busylight.lights.seriallight:available_lights:83 - MuteSync found 0
2025-02-24 14:26:31.513 | INFO     | busylight.lights.light:all_lights:149 - MuteSync found 0 lights.
2025-02-24 14:26:31.513 | INFO     | busylight.lights.light:all_lights:142 - Light found 1 lights total.
2025-02-24 14:26:31.608 | INFO     | busylight.lights.light:exclusive_access:528 - Entering exclusive access for Status Indicator
2025-02-24 14:26:31.609 | INFO     | busylight.lights.light:update:559 - data:10 = b'\x00\x00\x00\x00\x80\x08\x00\x80\xff"' wrote 9
2025-02-24 14:26:31.610 | INFO     | busylight.lights.light:exclusive_access:542 - Exiting exclusive access for Status Indicator
```

JnyJny, 2025-02-24:
Hi @EddiesTech thanks for the bug report and debug output.

```
2025-02-24 14:26:31.609 | INFO     | busylight.lights.light:update:559 - data:10 = b'\x00\x00\x00\x00\x80\x08\x00\x80\xff"' wrote 9
```

Not sure exactly what is going on yet, but the data is definitely missing a terminal 0x22 byte; 0xff22 are the final bytes of the command buffer. This is probably related to #417.  Especially since the command buffer length is 10 and the number of bytes written is 9.

EddiesTech, 2025-02-25:
No, it stays off.

EddiesTech, 2025-03-04:
I've just tested on another Windows PC and got the same result

JnyJny, 2025-03-04:
Well, honestly I'm struggling to figure this out. I have another script to run to gather some more information:

```
"""
"""
import sys
import platform
from time import sleep

import hid

def dump(buf: bytes) -> str:
    return str(len(buf)) + "|" + " ".join(f"{b:02X}" for b in buf)

system = platform.system()
uname = platform.uname()

try:
    which_cmd = sys.argv[1]
except IndexError:
    which_cmd = "9"
print(f"     Which: {which_cmd}")
for key,value in uname._asdict().items():
    print(f"{key:>10s}: {value}")

    
try:
    device_info = hid.enumerate(0x047F, 0xD005)[0]
    device = hid.device()
    device.open_path(device_info["path"])
except Exception as error:
    print(f"Failed to find Status Indicator {error}")
    exit(1)

match which_cmd:
    case "9":
        on_cmd = bytes( [0, 0, 0, 0x80, 0, 0, 0, 0xFF, 0x22])
        off_cmd = bytes([0, 0, 0, 0, 0, 0, 0, 0xFF, 0x22])
    case "10":
        on_cmd = bytes([0, 0, 0, 0, 0x80, 0, 0, 0, 0xFF, 0x22])
        off_cmd = bytes([0, 0, 0, 0, 0, 0, 0, 0, 0xFF, 0x22])    

for cmd in [on_cmd, off_cmd]:
    print(f"Cmd: {dump(cmd)}")
    nbytes = device.write(cmd)
    print(f"Sent {nbytes} bytes")
    sleep(1)
```

This script takes a command line argument, and expects either 9 or 10:
```console
> python3 script.py 9
...
> python3 script.py 10
...
```

I would appreciate it if you could run the script both ways and give me the results.

Thanks for your patience.

EddiesTech, 2025-03-04:
Hi, running with 9 works! Light goes green and then off. Running with 10 doesn't, stays blank. Not sure what is different, but here are the outputs. To be clear, top one works, bottom didn't.

[image]

JnyJny, 2025-03-04:
That's good news! Kinda. I guess I'll need to check which release of Windows instead of just Windows to determine when to append an extra byte. I should have an updated release shortly.

JnyJny, 2025-03-04:
I've published version 0.33.3 to PyPI, please give it a try and let me know if it's working for you.

EddiesTech, 2025-03-05:
That all works with this version. Thanks for the help :)

closed 2025-03-05 by EddiesTech