JnyJny/busylight #436: [BUG] kuando Path Error

spicywolff, 2025-03-23:
**Software Versions:**
 - Operating System: Linux Mint 22.1 Cinnamon (Ubuntu based)
 - Python version:  Python 3.12.3 (Virtual Env.)
 - BusyLight version: Newest kuando Omega

**General Type of Problem**
- [ X ] Command-line

**Describe the Problem**
The script / busylight (kuando omega) throws an path error

**Expected Behavior**
Light gets listed and is ready to work.

**Error Output**
```
(busylights) admeme@X280:~/Dokumente/busylights$ busylight --debug list
2025-03-23 11:26:11.251 | INFO     | busylight.__main__:precommand_callback:132 - version 0.33.3
2025-03-23 11:26:11.251 | INFO     | busylight.__main__:precommand_callback:133 - timeout=None
2025-03-23 11:26:11.251 | INFO     | busylight.__main__:precommand_callback:134 -     dim=1.0
2025-03-23 11:26:11.251 | INFO     | busylight.__main__:precommand_callback:135 -  lights=[]
2025-03-23 11:26:11.251 | INFO     | busylight.__main__:precommand_callback:136 -     cmd='list'
2025-03-23 11:26:11.252 | INFO     | busylight.__main__:list_available_lights:275 - listing connected lights
2025-03-23 11:26:11.252 | INFO     | busylight.lights.hidlight:available_lights:52 - BlinkStick found 0
2025-03-23 11:26:11.252 | INFO     | busylight.lights.light:all_lights:149 - BlinkStick found 0 lights.
2025-03-23 11:26:11.252 | INFO     | busylight.lights.hidlight:available_lights:52 - Blynclight found 0
2025-03-23 11:26:11.252 | INFO     | busylight.lights.light:all_lights:149 - Blynclight found 0 lights.
2025-03-23 11:26:11.253 | INFO     | busylight.lights.hidlight:available_lights:52 - Blynclight_Mini found 0
2025-03-23 11:26:11.253 | INFO     | busylight.lights.light:all_lights:149 - Blynclight_Mini found 0 lights.
2025-03-23 11:26:11.253 | INFO     | busylight.lights.hidlight:available_lights:52 - Blynclight_Plus found 0
2025-03-23 11:26:11.253 | INFO     | busylight.lights.light:all_lights:149 - Blynclight_Plus found 0 lights.
2025-03-23 11:26:11.253 | INFO     | busylight.lights.hidlight:available_lights:52 - Status_Indicator found 0
2025-03-23 11:26:11.253 | INFO     | busylight.lights.light:all_lights:149 - Status_Indicator found 0 lights.
2025-03-23 11:26:11.254 | INFO     | busylight.lights.hidlight:available_lights:52 - Busylight_Alpha found 0
2025-03-23 11:26:11.254 | INFO     | busylight.lights.light:all_lights:149 - Busylight_Alpha found 0 lights.
2025-03-23 11:26:11.254 | INFO     | busylight.lights.hidlight:available_lights:52 - Busylight_Omega found 1
2025-03-23 11:26:11.254 | ERROR    | busylight.lights.hidlight:acquire:111 - open_path failed: open failed
2025-03-23 11:26:11.254 | ERROR    | busylight.lights.light:all_lights:148 - Busylight_Omega b'1-1:1.0'
2025-03-23 11:26:11.254 | INFO     | busylight.lights.light:all_lights:149 - Busylight_Omega found 0 lights.
2025-03-23 11:26:11.255 | INFO     | busylight.lights.hidlight:available_lights:52 - Flag found 0
2025-03-23 11:26:11.255 | INFO     | busylight.lights.light:all_lights:149 - Flag found 0 lights.
2025-03-23 11:26:11.255 | INFO     | busylight.lights.hidlight:available_lights:52 - Bluetooth found 0
2025-03-23 11:26:11.255 | INFO     | busylight.lights.light:all_lights:149 - Bluetooth found 0 lights.
2025-03-23 11:26:11.255 | INFO     | busylight.lights.hidlight:available_lights:52 - Mute found 0
2025-03-23 11:26:11.255 | INFO     | busylight.lights.light:all_lights:149 - Mute found 0 lights.
2025-03-23 11:26:11.256 | INFO     | busylight.lights.hidlight:available_lights:52 - Orb found 0
2025-03-23 11:26:11.256 | INFO     | busylight.lights.light:all_lights:149 - Orb found 0 lights.
2025-03-23 11:26:11.256 | INFO     | busylight.lights.hidlight:available_lights:52 - MuteMe found 0
2025-03-23 11:26:11.256 | INFO     | busylight.lights.light:all_lights:149 - MuteMe found 0 lights.
2025-03-23 11:26:11.256 | INFO     | busylight.lights.hidlight:available_lights:52 - MuteMe_Mini found 0
2025-03-23 11:26:11.256 | INFO     | busylight.lights.light:all_lights:149 - MuteMe_Mini found 0 lights.
2025-03-23 11:26:11.257 | INFO     | busylight.lights.hidlight:available_lights:52 - Blink1 found 0
2025-03-23 11:26:11.257 | INFO     | busylight.lights.light:all_lights:149 - Blink1 found 0 lights.
2025-03-23 11:26:11.265 | INFO     | busylight.lights.seriallight:available_lights:83 - BusyTag found 0
2025-03-23 11:26:11.265 | INFO     | busylight.lights.light:all_lights:149 - BusyTag found 0 lights.
2025-03-23 11:26:11.270 | INFO     | busylight.lights.seriallight:available_lights:83 - Fit_StatUSB found 0
2025-03-23 11:26:11.270 | INFO     | busylight.lights.light:all_lights:149 - Fit_StatUSB found 0 lights.
2025-03-23 11:26:11.277 | INFO     | busylight.lights.seriallight:available_lights:83 - MuteSync found 0
2025-03-23 11:26:11.277 | INFO     | busylight.lights.light:all_lights:149 - MuteSync found 0 lights.
2025-03-23 11:26:11.277 | INFO     | busylight.lights.light:all_lights:142 - Light found 0 lights total.
No lights detected.
```

JnyJny, 2025-03-23:
Thank you for the bug report. I'm still scratching my head about what's going on. The path that is causing the error comes from hidapi ( or possibly pyhidapi ). Can you confirm which of these packages are in your virtual environment?

```console
$ python -m pip list | grep -i hid
```

My dev environment sees:
```console
$ python3 -m pip list | grep -i hid
hidapi               0.14.0.post4
```

It would be very helpful if you could run this small script and post the results:

```python
import hid
from pprint import pprint

devices = hid.enumerate(0x27BB,0)

for device in devices:
    pprint(device)
```

spicywolff, 2025-03-23:
Hi, the problem is I always have to create a new Virtual Env. and I am not a Coder. I use Mint.

This is the Output to the first:
```
(venv) admeme@X280:~/Dokumente/busylights$ python -m pip list | grep -i hid
hidapi               0.14.0.post4
```

This is the Output from the second:

```
(venv) admeme@X280:~/Dokumente/busylights$ python3 hid.py 
Traceback (most recent call last):
  File "/home/admeme/Dokumente/busylights/hid.py", line 1, in <module>
    import hid
  File "/home/admeme/Dokumente/busylights/hid.py", line 4, in <module>
    devices = hid.enumerate(0x27BB,0)
              ^^^^^^^^^^^^^
AttributeError: partially initialized module 'hid' has no attribute 'enumerate' (most likely due to a circular import)
```

and the third I really do not understand what I need to do :(

JnyJny, 2025-03-23:
No worries, you don't have to be a coder to help me out.  Rename the file from `hid.py` to `test.py` and re-run it:

```console
$ mv hid.py test.py
$ python3 test.py
```

spicywolff, 2025-03-24:
> No worries, you don't have to be a coder to help me out. Rename the file from `hid.py` to `test.py` and re-run it:
> 
> $ mv hid.py test.py
> $ python3 test.py

```
admeme@X280:~/Dokumente/busylights$ python3 test.py 
Traceback (most recent call last):
  File "/home/admeme/Dokumente/busylights/test.py", line 1, in <module>
    import hid
ModuleNotFoundError: No module named 'hid'
```

Hope this helps solving this Error

spicywolff, 2025-03-24:
> We are getting closer!
> 
> You'll need to activate the virtual environment you created for BusyLight and then run the `test.py` script:
> 
> $ cd path/to/busylight
> $ source venv/bin/activate
> <busylight> $ python3 test.py
> ...

Here is the output:
```

(venv) admeme@X280:~/Dokumente/busylights$ python3 test.py 
{'bus_type': 1,
 'interface_number': 0,
 'manufacturer_string': '',
 'path': b'1-1:1.0',
 'product_id': 15311,
 'product_string': '',
 'release_number': 256,
 'serial_number': '',
 'usage': 0,
 'usage_page': 0,
 'vendor_id': 10171}
```

JnyJny, 2025-03-24:
Great, this is a step in the right direction. Unfortunately, it's not enough information to solve this problem yet. It's strange that so much of the data is not populated in the output from your machine (vs the output generated for my Busylight Omega).

I don't have a Mint installation to look at, but that path value is very suspect.  Could you run this command and post the results?

```console
$ lsusb -v
```

Also, did you [install any udev rules](https://github.com/JnyJny/busylight?tab=readme-ov-file#linux-post-install-activities) before trying to use your device?  If you did, could you please post the contents of the udev rules you added?

spicywolff, 2025-03-26:
> Great, this is a step in the right direction. Unfortunately, it's not enough information to solve this problem yet. It's strange that so much of the data is not populated in the output from your machine (vs the output generated for my Busylight Omega).
> 
> I don't have a Mint installation to look at, but I get very similar paths on a Suse based machine. Could you run this command and post the results?
> 
> $ lsusb -v
> 
> Also, did you [install any udev rules](https://github.com/JnyJny/busylight?tab=readme-ov-file#linux-post-install-activities) before trying to use your device? If you did, could you please post the contents of the udev rules you added?

Sorry for the late response, here is the output:

```
(venv) admeme@X280:~/Dokumente/busylights$ lsusb -v

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            6.08
  iManufacturer           3 Linux 6.8.0-56-generic xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:00:14.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12

Bus 001 Device 002: ID 046d:c53f Logitech, Inc. USB Receiver
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
[... 1766 lines cut ...]
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0

```

spicywolff, 2025-03-26:
Oh and no DevRules, no.

JnyJny, 2025-03-26:
So given all the messages:
```
Couldn't open device, some information will be missing
```

in the `lsusb` output, I think it's a permissions problem that will be solved by installing `udev` rules (at least for the Busylight Omega, the udev rules I provide don't work for other device).

spicywolff, 2025-03-29:
> So given all the messages:
> 
> ```
> Couldn't open device, some information will be missing
> ```
> 
> in the `lsusb` output, I think it's a permissions problem that will be solved by installing `udev` rules (at least for the Busylight Omega, the udev rules I provide don't work for other device).

Sorry, I cant test it rn. I'm moving and it's a hassle. What are uDev Rules?

JnyJny, 2025-03-29:
The udev subsystem is a device manager for the Linux kernel. In our particular case, we are interested in the rules that udev uses when USB devices are plugged in and unplugged. By default, udev will only give a limited amount of access to devices for non-root users. All of the commands and scripts I had you run appeared to have been run by non-root user (you), and they all failed with similar errors. Writing udev rules can be a little tricky but `busylight` has a subcommand that will print out udev rules for all the devices it knows about. The directions for printing out the udev rules and installing them can be found in the README.md for this project, [here](https://github.com/JnyJny/busylight?tab=readme-ov-file#linux-post-install-activities). I suspect once you have those installed you will have more success with `busylight`.

spicywolff, 2025-04-01:
> The udev subsystem is a device manager for the Linux kernel. In our particular case, we are interested in the rules that udev uses when USB devices are plugged in and unplugged. By default, udev [...]

I go insane, worked with UDev 🤦

closed 2025-04-01 by spicywolff