JnyJny/busylight #437: Windows 11 support

0ssiwald, 2025-04-19:
Hi, 
I had problems getting my busylight from kuando running on windows 11 and it seemed like I just had to add it in the list in the lights.py file (Line  554):

if self.platform in ["Windows_10", "Windows_11"]:

With this change everything worked fine. Can you add Windows 11 support for the next release?
Thank you for your nice work!

Best Regards,
Oswald

JnyJny, 2025-05-07:
Thank you for the bug report! I'm just now getting caught up and didn't realize this issue had been in the queue for so long!

I'm working on this and #439 which are likely the same problem and should have something for you to try shortly.

JnyJny, 2025-05-07:
Could you please provide the results of this python command?

```console
python -m platform
```

0ssiwald, 2025-05-07:
Windows-11-10.0.26100-SP0

closed 2025-06-05 by JnyJny