JnyJny/busylight #439: [BUG]

aracat, 2025-05-06:
**Software Versions:**
 - Operating System: Windows 23H2 22631.5189
 - Python version: 3.9.1
 - BusyLight version: 033.3

**General Type of Problem**
- [ ] Installation
- [ ] Command-line
- [ ] Integrating into another project
- [ ] Web API
- [x] other

**Describe the Problem**
I appear to have the opposite problem of https://github.com/JnyJny/busylight/issues/417 with a Blynclight device id = (0x2C0D, 0x000C) where the extra padding byte on Windows is corrupting the command:

light.on((255,0,0)) -> should be bright red, is bright blue (this model's command structure is R B G Flags ...)
light.on((255,0,255)) -> should be bright purple, is bright cyan
light.on((255,255,255)) -> should be bright white, is off (due to the green byte getting shifted into the flags and setting the `off` bit)

After editing light.py to delete the addition of the padding byte, the device functions as expected.

**Error Output**
If there was an error displayed, paste it in here.

JnyJny, 2025-05-07:
Thank you for the bug report and the detailed teardown of how it's failing!  Getting Windows right with respect to I/O has been a challenge for me. The environmental cues like `platform.system` and friends have not led me to a stable solution where I _know_ when to prepend an extra byte or not.  Could you provide the results of this command on your machine?

```console
python -m platform
```

aracat, 2025-05-08:
> Thank you for the bug report and the detailed teardown of how it's failing! Getting Windows right with respect to I/O has been a challenge for me. The environmental cues like `platform.system` and friends have not led me to a stable solution where I _know_ when to prepend an extra byte or not. Could you provide the results of this command on your machine?
> 
> python -m platform

I have two Windows 11 systems available, both of which exhibit this issue. Their outputs are:

`Windows-10-10.0.22631-SP0` and `Windows-10-10.0.22621-SP0`

closed 2025-06-05 by JnyJny